ghdl/ghdl #647: Support custom types in VCD waveforms

darkdragon-001, 2018-09-09:
Please support custom types in `.vcd` waveforms. Your own `.ghw` waveforms do support this.

Example code:

    TYPE state_type is (s1, s2, s3);
    SIGNAL state, next_state :state_type;

Paebbels, 2018-09-09:
The [*Value Change Dump* (VCD)](https://en.wikipedia.org/wiki/Value_change_dump) format is limited to the elements as described in *IEEE Std. 1364–1995*, because it is defined as part of the [Verilog language](https://en.wikipedia.org/wiki/Verilog).

I'm not aware of any updates to the VCD format since then.

darkdragon-001, 2018-09-09:
Seems that there is indeed no enum support in Verilog (only in SystemVerilog). 

Maybe it still makes sense to _enumerate_ the values (via `state_type'POS(state)`) and export them as integer in the VCD file.