ghdl/ghdl #648: internal compiler error: in expand_expr_real_1, at expr.c:10001

vicencb, 2018-09-09:
Hi,
this VHDL description:
```vhdl
entity e is end entity;
library ieee;
architecture h of e is
  constant L :positive := integer(ieee.math_real.ceil(9.9));
  signal   s :bit_vector(31 downto 0);
  alias    a :bit_vector(L-1 downto 2) is s(L-1 downto 2);
  signal   b :bit_vector(L-1 downto 2);
begin
  a <= b;
end architecture;
```
makes GHDL to crash with this message:
```
during RTL pass: expand
/tmp/ice.vhd: In function 'work__e__ARCH__h__DECL_ELAB':
/tmp/ice.vhd:6: internal compiler error: in expand_expr_real_1, at expr.c:10001
   alias    a :bit_vector(L-1 downto 2) is s(L-1 downto 2);
 
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
ghdl:error: exec error
```
when executing:
```sh
ghdl -a ice.vhd
```
with GHDL version:
```
GHDL 0.36-dev (v0.35-251-g258bbf95) [Dunoon edition]
 Compiled with GNAT Version: 8.2.0
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
when running on the arm64 architecture.

Regards,
&nbsp;&nbsp;Vicenç.

diogratia, 2018-09-10:
I ran through a boatload of behavioral tests to try and determine what's going on here. It appear to be the case of an uninitialized pointer evaluating memory. You can change all most anything and get the symptom to go away. (For instance, commenting out the signal assignment).  

I verified the original problem with the llvm back end getting a distinct error message using a command line analyzer command:

```bash
ghdl -a e.vhdl

Execution terminated by unhandled exception
Exception name: CONSTRAINT_ERROR
Message: erroneous memory access
Load address: 0x1030b2000
Call stack traceback locations:
0x103e00527 0x7fff96cde528
ghdl:error: compilation error
```
and
```bash
ghdl --version
GHDL 0.36-dev (v0.35-271-g5cd3de4) [Dunoon edition]
 Compiled with GNAT Version: GPL 2015 (20150428-49)
 llvm code generator
```
Comment out the signal assignment:

```vhd
  -- a <= b;
```

and 

```bash
ghdl -a e.vhdl
```
(no error).

Where you can then elaborate and run with no error (not too surprising).

tgingold, 2018-09-12:
I can reproduce the crash.  It is a particular case; but that should be fixable...

tgingold, 2018-09-18:
Now fixed.

closed 2018-09-18 by tgingold

vicencb, 2018-09-18:
I can confirm it now works with version v0.35-300-g262c190b.
By the way: very useful addition of "declaration X hides Y".
Thanks!