ghdl/ghdl #652: Do library names and signal names share the same namespace?

Paebbels, 2018-09-15:
**Description**
When using direct entity instantiation, GHDL is complaining about a prefix of a record, that is not a signal.

> C:/Temp/ghdl/ClockIsNotARecord.vhdl:26:26: signal "clock" does not designate a record

I have used the name `clock` for:
- the **library name**  
   see `--work=clock` below in the shell script
- a **package name**  
   referencing `clock.clock.all` seems to be fine.
- a **local signal** in my testbench  
   no error while declaring it

**Expected behavior**
In that particular case, only a library name can be used. No signal names are allowed, right.

1. If both (or all three) names collide, an error should be generated, when signal `clock` is declared.  
2. If a name is hidden:
   * a warning should be issued when signal `clock` is generated, and 
   * better error message should be generated when the entity is instantiated.

**Context**
```
$ ghdl -a --std=08 --work=clock /c/Temp/ghdl/ClockIsNotARecord.vhdl
C:/Temp/ghdl/ClockIsNotARecord.vhdl:26:26: signal "clock" does not designate a record
C:/Temp/ghdl/ClockIsNotARecord.vhdl:31:26: signal "clock" does not designate a record
```

- OS: Windows
- Origin:
  - [x] Built from sources.  
     Latest GHDL 0.36

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

**How to reproduce?**

```
#>> ent.vhd
entity clock_Generator is
	port (
		Clock : out bit
	);
end entity;

entity clock_Monitor is
	port (
		Clock : in bit
	);
end entity;

package clock is
end package;

library Clock;
use     Clock.clock.all;

entity test is
end entity;

architecture tb of test is
  signal Clock        : bit;
begin
	gen: entity Clock.clock_Generator
		port map (
			Clock       => Clock
		);
	
	mon: entity Clock.clock_Monitor
		port map (
			Clock       => Clock
		);
end architecture;

#>> sim.sh
ghdl -a --std=08 --work=clock ent.vhd

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

Paebbels, 2018-09-15:
This error is not emitted by Riviera-PRO.

tgingold, 2018-09-16:
There is no namespace in vhdl; the error is correctly detected.

What error message would you like ?

Paebbels, 2018-09-16:
Compilers have name scopes and some can also distinguish between different entities in the same scope. Isn't that concept called *namespace*? I think VHDL calls it visibility?

E.g. library `clock` is in a global scope and by using `library clock;` a package is made also visible in the global scope, right?

When declaring the signal `clock`, an object is also made visible. As far as I can see, having these three names is not a problem for VHDL, otherwise it would stop at the signal declaration line.

So when writing `gen: entity Clock.clock_Generator`, shouldn't an instantiation filter visible names only for library names, because only a library name is allowed here? I think this is what Riviera-PRO is doing.

go2sh, 2018-09-16:
No. With the signal declaration, you hide the library declaration. (See LRM 12.3)

> A declaration is said to be hidden within (part of) an inner declarative region if the inner region contains a homograph of this declaration; the outer declaration is then hidden within the immediate scope of the inner homograph. Each of two declarations is said to be a homograph of the other if and only if both declarations have the same designator, and they denote different named entities, and either overloading is allowed for at most one of the two, or overloading is allowed for both declarations and they have the same parameter and result type profile (see 4.5.1).

Your library is simply not visible any more within your architecture. Only the signal is visible. VHDL in the first place doesn't care about the type of a named entity. There is no such thin as a filter of names depending on the construct you use it in.

go2sh, 2018-09-16:
> No. With the signal declaration, you hide the library declaration. (See LRM 12.3)
>
>
> Your library is simply not visible any more within your architecture. Only the signal is visible. VHDL in the [...]

PS.: I'am not 100% sure though.^^ This part of the LRM is hard to read.

Paebbels, 2018-09-16:
So, if the signal is hiding the library (and package?), there is no way to reference the library anymore? Normally, we could use selected names to create a fully qualified name (FQN) like we use for `std.env.stop`.

If this is true, I would argue, that GHDL should emit a warning, whenever a name is created that hides another name and there is no way to use FQN anymore.

**Reading your quote:**  
I'm not sure if the definition of homographs apply to library names. I cannot find, that a library is a named entity in VHDL.

**What about 12.3 Visibility?**
> [...]  
> A declaration is visible only within a certain part of its scope; this part starts at the end of the declaration except in the declaration of a design unit other than a PSL verification unit, a package declaration, or a protected type declaration, in which case it starts immediately after the reserved word is occurring after the identifier of the design unit, package declaration, or protected type declaration. This rule applies to both explicit and implicit declarations.  
Visibility is either by selection or direct. A declaration is visible by selection at places that are defined as follows:
> 1. **For a primary unit contained in a library: at the place of the suffix in a selected name whose prefix denotes the library.**
> 2. For an entity name in a configuration declaration whose entity name is a simple name: at the place of the simple name, and the context is that of the library WORK.
> 3. [...]

Paebbels, 2018-09-16:
In *7.3.3 Default binding indication*, the LRM writes about visibile entities, but not about visibility when it comes to the form `L.C` wherein `L` is a library and `C` a component name.

> **7.3.3 Default binding indication**  
> In certain circumstances, a default binding indication will apply in the absence of an explicit binding indication. The default binding indication consists of a default entity aspect, together with a default generic map aspect and a default port map aspect, as appropriate.
>
> If no visible entity declaration has the same simple name as that of the instantiated component, then the default entity aspect is open. A visible entity declaration is the first entity declaration, if any, in the following list:
> 
> 1. An entity declaration that has the same simple name as that of the instantiated component and that is directly visible (see Clause 12.3),
> 2. An entity declaration that has the same simple name as that of the instantiated component and that would be directly visible in the absence of a directly visible (see Clause 12.3) component declaration with the same simple name as that of the entity declaration, or
> 3. **An entity declaration denoted by L.C, where L is the target library and C is the simple name of the instantiated component. The target library is the library logical name of the library containing the design unit in which the component C is declared.**

go2sh, 2018-09-16:
> So, if the signal is hiding the library (and package?), there is no way to reference the library anymore? Normally, we could use selected names to create a fully qualified name (FQN) like we use for `std.env.stop`.
I would say yes.
> If this is true, I would argue, that GHDL should emit a warning, whenever a name is created that hides another name and there is no way to use FQN anymore.
Maybe in a debug mode or something, but the language is very clear about that. The generale case of 12.3 is used by a lot of people and the same as in other languages.
> I'm not sure if the definition of homographs apply to library names. I cannot find, that a library is a named entity in VHDL.
I would argue the other way around. There is no special treatment for libraries in a selected name, a library as prefix is just a simple name.

I think both your examples require the library to be visible. Your examples are one step ahead of what this discussion is about, weather a library is a named entity and if it can be hidden by homographs.

Paebbels, 2018-09-16:
According to ModelSim 10.5b (Oct. 2016), GHDL is *right*.

It is reporting this error:
> ** Error: C:\Temp\ghdl\ClockIsNotARecord.vhdl(26): Unknown expanded name.

**ModelSim 10.5b outputs:**
```
vcom -2008 -work clock C:\Temp\ghdl\ClockIsNotARecord.vhdl
-- Loading package STANDARD
-- Compiling entity clock_Generator
-- Compiling entity clock_Monitor
-- Compiling package clock
-- Loading package clock
-- Compiling entity test
-- Compiling architecture tb of test
** Error: C:\Temp\ghdl\ClockIsNotARecord.vhdl(26): Unknown expanded name.
** Error: C:\Temp\ghdl\ClockIsNotARecord.vhdl(31): Unknown expanded name.
** Error: C:\Temp\ghdl\ClockIsNotARecord.vhdl(35): VHDL Compiler exiting
End time: 20:19:19 on Sep 16,2018, Elapsed time: 0:00:00
Errors: 3, Warnings: 0
```

tgingold, 2018-09-16:
I can add a warning when a declaration is hidden.

Paebbels, 2018-09-16:
I did another test with GHDL and ModelSim.

1. I change the clock signal to `clock1`, so both simulators can analyze without problems.  
    Signal `clock` is no longer hiding the library name.
2. Next, I added a constant and a signal to the package:
   ```vhdl
     constant clock_Generator : integer := 5;
     signal   clock_Monitor   : integer := 10;
    ```
3. I added a signal assignment to the architecture, referring to the global signal `clock.clock_Monitor` from package `clock`.  
    ```vhdl
    clock.clock_Monitor <= 15;
    ```

The name `clock.clock_Generator` can now refer to a library + entity or to a package + constant.

* **For 2)**:  
   Both tools can distinguish this and choose the library name over the package name.
* **For 3)**:  
   Both tools reject to assign to an entity name.

Is this behavior intended by VHDL?  
I was believing that every element can be chosen if an appropriate long name is used.

Paebbels, 2018-09-16:
> I can add a warning when a declaration is hidden.

I think this would be best.

I don't want a warning for every hidden object, because most would be intended e.g. in nested loops. But an object hidden in a way it gets unreachable should be a warning.

go2sh, 2018-09-16:
The package clock is not visible. Even it is in the same file. You have to make package explicitly visible with a use clause or select it from the library clock. I'am not sure what happens if you do:
```
library clock;
use clock.clock;
```
I would think, the package would not be made visible, since the library already denote a name with clock. This is also the current behavior in ghdl

Paebbels, 2018-09-16:
Using

```vhdl
clock.clock.clock_Monitor <=15;
```

is accepted by both tools.
Note: Signal `clock` was renamed to `clock1`.

closed 2018-09-19 by tgingold

Paebbels, 2018-09-23:
I did ne new compilation of GHDL under MinGW64 for testing.
The new compilation of the IEEE packages generated a lot of "declaration of ... hides ..." messages :).