ghdl/ghdl #655: Discrepancy between stop-time and NOW

Paebbels, 2018-09-15:
This is a followup issue to #654. This issues describes what is going wrong, but actually, I'm not able to create an MCVE. In a bigger testbench I can produce the `NOW is` line, but not in an MCVE.

Any ideas, how the mentioned line is generated? It's not from OSVVM, because simulation is stopped at that time. Moreover, OSVVM (VHDL code) has no access to the current delta-cycle number, right?

--------------
**Description**
The time emitted by NOW does not match the time when the simulation was stopped by `std.env.stop`.

> simulation stopped @0ms with status 1
> Now is 9223372036854775807fs +0

**Expected behaviour**
The time emitted by NOW should be `0 fs`, but not `time'high`.

**Context**
```
Linking in memory
Starting simulation
%% Alert FAILURE in generator clock_Generator, PeriodLow has no positive value. at 0 ns

%% Alert Stop Count on FAILURE reached in generator clock_Generator at 0 ns
%% DONE  FAILED  AlertLogTop  Total Error(s) = 1  Failures: 1  Errors: 0  Warnings: 0  at 0 ns
%%    Default                    Failures: 0  Errors: 0  Warnings: 0
%%    OSVVM                      Failures: 0  Errors: 0  Warnings: 0
%%    generator clock_Generator  Failures: 1  Errors: 0  Warnings: 0
simulation stopped @0ms with status 1
Now is 9223372036854775807fs +0
```

- OS: Windows 10
- Origin:
  - [x] Built from sources.  
    GHDL 0.36-dev

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

**How to reproduce?**
*The GHDL commands have been adapted to match the requirements for the Docker image.*

```vhdl
#>> ent.vhd
entity test is
end entity;

architecture tb of test is
begin
  std.env.stop(1);
end architecture;

#>> sim.sh
ghdl -a --std=08 ent.vhd
ghdl -e --std=08 test
ghdl -r --std=08 -v test --disp-time --stop-time=100

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [ ] Add a MWE
- [x] Try the latest version

diogratia, 2018-09-16:
When there are no more signal updates scheduled for a future time in any projected output waveform queues simulation time is advance to to TIME'HIGH (See IEEE Std 1076-2008 14.7.5.3 Simulation cycle H) 5), 14.7.5.1 General para 2). By a not so amazing coincidence  in an implementation dependent ghdl (as shown by `ghdl --disp-standard`) TIME'HIGH is 9223372036854775807 primary units (fs).

The command line --stop-time isn't specified to continue until simulation is complete (at TIME'HIGH) while STOP does. (--stop-time is analogous to FINISH).

The functionality of stop is specified in 16.5 Standard environment package:

>Execution of the STOP procedures causes the same action by the host simulator as that caused by the vhpi_control function called with the argument vhpiStop (see 23.5). Execution of the FINISH procedures causes the same action by the host simulator as that caused by the vhpi_control function called with the argument vhpiFinish (see 23.5). Execution shall not return to the VHDL description after a call to the FINISH procedure. For the procedures with the STATUS parameter, the value of the STATUS parameter may be used in an implementation defined manner by the host simulator. For the procedures with no parameter, the effect is the same as that caused by the vhpi_control function with no additional arguments beyond the vhpiStop or vhpiFinish argument.

23.5 vhpi_control:

>This standard specifies three control actions, corresponding to the enumeration values vhpiStop, vhpiFinish, and vhpiReset of type vhpiSimControlT. If a tool implements any of these control actions, the effect shall be as follows:  
>>— If command is vhpiStop, after control returns to the tool from the callback function from which the vhpi_control function was invoked, the tool stops simulation then accepts further directives from an interactive user or a command source. Additional arguments to vhpi_control beyond the command argument may be interpreted by the tool in an implementation-defined manner. The tool shall provide an implementation-defined default action if no additional arguments are provided.  
>>  
>>— If command is vhpiFinish, after control returns to the tool from the callback function from which the vhpi_control function was invoked, the tool enters the termination phase (see 20.10). Additional arguments to vhpi_control beyond the command argument may be interpreted by the tool in an implementation-defined manner. The tool shall provide an implementation-defined default action if no additional arguments are provided.  
>>  
>>...

And we see the Stop command executes a normal tool simulation stop while Finish  terminates the simulation.  A simulation normally always ends at TIME'HIGH.

tgingold, 2018-09-16:
Can you post the full testbench ?