ghdl/ghdl #656: CONSTRAINT_ERROR

Ravenwater, 2018-09-15:
**CONSTRAINT_ERROR**

**Expected behaviour**
concurrent signal assignments.

**Context**
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: 0.35 (tarball) [Dunoon edition]
Compiled with GNAT Version: GPL 2017 (20170515-63)
Target: i686-pc-mingw32
In directory: C:\Users\tomtz\Documents\dev\clones\posit-data-path-generator\
Command line:
C:\eda\ghdl\bin\ghdl.exe -r --std=08 p8e0_add_tb behavior --wave=posit_add.ghw
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-rtis_utils.adb:261 invalid data
******************************************************************

**Additional context**
How do I get more information about what triggers this error? 

I am using generic packages, and first thought is that the generics do not propagate to the VHDL code, but without more help what structure is being simulated, unclear how to correlate that hunch.

tgingold, 2018-09-16:
You just need to post a reproducer; or even the whole design if that's easier.

Ravenwater, 2018-09-16:
Isolated it to be related to the argument --wave=posit_add.ghw, that is, removing the wave generation yields a working simulator.

Still working on getting a bug reproduction example for testing....

Paebbels, 2018-09-16:
You're using GHDL 0.35. The latest version is 0.36-dev. There have been some updates to the the output code since 0.35. I'll compile a new Windows mcode version for you ...

Ravenwater, 2018-09-16:
here is a reproduction:

$ sh repro.sh

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: 0.35 (tarball) [Dunoon edition]
Compiled with GNAT Version: GPL 2017 (20170515-63)
Target: i686-pc-mingw32
In directory: C:\Users\tomtz\Documents\dev\clones\scratch\constraint_error_repro\
Command line:
C:\eda\ghdl\bin\ghdl.exe -r --std=08 tb behavior --wave=test.ghw
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-rtis_utils.adb:261 invalid data
******************************************************************

tomtz@SW-LAPTOP-250 MINGW64 /C/Users/tomtz/Documents/dev/clones/scratch/constraint_error_repro (master)
-- removing the --wave=test.ghw
$ sh repro.sh
./tb.vhdl:31:16:@0ms:(report note): NBITS = 8
./tb.vhdl:32:16:@0ms:(report note): EBITS = 0
./tb.vhdl:33:16:@0ms:(report note): SBITS = 5
./tb.vhdl:34:16:@0ms:(report note): FBITS = 5
./tb.vhdl:36:16:@0ms:(report note): GUARDBITS = 3
./tb.vhdl:39:16:@0ms:(report note): BITS_ADDER_OUT      : 7
./tb.vhdl:40:16:@0ms:(report note): BITS_MULTIPLIER_OUT : 11
./tb.vhdl:41:16:@0ms:(report note): BITS_DIVIDER_OUT    : 13
./tb.vhdl:42:16:@0ms:(report note): BITS_MAX_ARITH_OUT  : 13
./test_unit.vhdl:94:24:@0ms:(report note): start fraction bits = 5
./tb.vhdl:47:16:@10ns:(report note): isZero : 1
./tb.vhdl:51:16:@15ns:(report note): isNaR  : 1
./test_unit.vhdl:94:24:@15ns:(report note): start fraction bits = 5
./tb.vhdl:55:16:@20ns:(report note): isZero : 0
./tb.vhdl:56:16:@20ns:(report note): isNaR  : 0
./tb.vhdl:57:16:@20ns:(report note): Sign   : 1
./tb.vhdl:58:16:@20ns:(report note): Scale  : 0
./tb.vhdl:59:16:@20ns:(report note): Fract  : 00000

files ----------------------------------------------------------
repro.sh
$ cat repro.sh
```
#export OPTIONS="--ieee=synopsys --std=08"
export OPTIONS="--std=08"

# reusable functions
#ghdl -a $OPTIONS ./tb_pkg.vhdl

# interface packages
ghdl -a $OPTIONS ./repro_pkg.vhdl

# entity/architecture/testbench components
ghdl -a $OPTIONS ./test_unit.vhdl
ghdl -a $OPTIONS ./tb.vhdl

# elaborate the design
ghdl -e $OPTIONS tb

# run the design
# ghdl -r [OPTS] UNIT [ARCH] [RUNOPTS]  Run UNIT
ghdl -r $OPTIONS tb behavior --wave=test.ghw
#ghdl -r $OPTIONS tb behavior

# display the results
#gtkwave -f test.ghw debug_test.gtkw
```
$ repro_pkg.vhdl
```
library ieee;
[... 199 lines cut ...]
		report "BITS_DIVIDER_OUT    : " & to_string(BITS_DIVIDER_OUT);
		report "BITS_MAX_ARITH_OUT  : " & to_string(BITS_MAX_ARITH_OUT);
		wait for 5 ns;

		tb_raw <= "00000000";
		wait for 5 ns;
		report "isZero : " & to_string(tb_ssf.isZero);

		tb_raw <= "10000000";
		wait for 5 ns;
		report "isNaR  : " & to_string(tb_ssf.isNaR);

		tb_raw <= "11000000";
		wait for 5 ns;
		report "isZero : " & to_string(tb_ssf.isZero);
		report "isNaR  : " & to_string(tb_ssf.isNaR);
		report "Sign   : " & to_string(tb_ssf.sign);
		report "Scale  : " & to_string(to_integer(signed(tb_ssf.scale)));
		report "Fract  : " & to_string(tb_ssf.fraction);
		wait;
	end process;

end architecture;

```

Paebbels, 2018-09-16:
While compiling, I found #657, so I reverted my GNAT GPL setup and compiled with GNAT GPL 2017.

Extract GHDL e.g. into `C:\tools\GHDL\0.36-dev`.

----------------
**Attachments:**
* [ghdl-mcode-0.36-dev.zip](https://github.com/ghdl/ghdl/files/2387052/ghdl-mcode-0.36-dev.zip)

diogratia, 2018-09-16:
Your `repro.sh` is missing the architecture specifier in the elaboration command to match the run command:
```sh
# elaborate the design
ghdl -e $OPTIONS tb behavior # added architecture
```
After adding that for a more recent ghdl implementation there is no constraint error.

```none
ghdl --version
GHDL 0.36-dev (v0.35-271-g5cd3de4) [Dunoon edition]
 Compiled with GNAT Version: GPL 2015 (20150428-49)
 llvm code generator
```
commit 5cd3de4 is from ~ 24 August 2018 depending on the international dateline.

`repro.sh` finishes without error:

```
repro.sh
#!/bin/sh -v
#export OPTIONS="--ieee=synopsys --std=08"
export OPTIONS="--std=08"

# reusable functions
#ghdl -a $OPTIONS ./tb_pkg.vhdl

# interface packages
ghdl -a $OPTIONS ./repro_pkg.vhdl

# entity/architecture/testbench components
ghdl -a $OPTIONS ./test_unit.vhdl
ghdl -a $OPTIONS ./tb.vhdl

# elaborate the design
ghdl -e $OPTIONS tb behavior # added architecture

# run the design
# ghdl -r [OPTS] UNIT [ARCH] [RUNOPTS]  Run UNIT
ghdl -r $OPTIONS tb behavior --wave=test.ghw
#ghdl -r $OPTIONS tb behavior

# display the results
#gtkwave -f test.ghw debug_test.gtkw
```

noting test.ghw appears empty:
```
-rw-------  1 diogratia      staff  414 Sep 17 09:42 test.ghw
```
You could note one peculiarity:

```
>% ghdl -r tb behavior
./tb.vhdl:31:16:@0ms:(report note): NBITS = 8
./tb.vhdl:32:16:@0ms:(report note): EBITS = 0
./tb.vhdl:33:16:@0ms:(report note): SBITS = 5
./tb.vhdl:34:16:@0ms:(report note): FBITS = 5
./tb.vhdl:36:16:@0ms:(report note): GUARDBITS = 3
./tb.vhdl:39:16:@0ms:(report note): BITS_ADDER_OUT      : 7
./tb.vhdl:40:16:@0ms:(report note): BITS_MULTIPLIER_OUT : 11
./tb.vhdl:41:16:@0ms:(report note): BITS_DIVIDER_OUT    : 13
./tb.vhdl:42:16:@0ms:(report note): BITS_MAX_ARITH_OUT  : 13
./test_unit.vhdl:94:24:@0ms:(report note): start fraction bits = 5
./tb.vhdl:47:16:@10ns:(report note): isZero : 1
./tb.vhdl:51:16:@15ns:(report note): isNaR  : 1
./test_unit.vhdl:94:24:@15ns:(report note): start fraction bits = 5
./tb.vhdl:55:16:@20ns:(report note): isZero : 0
./tb.vhdl:56:16:@20ns:(report note): isNaR  : 0
./tb.vhdl:57:16:@20ns:(report note): Sign   : 1
./tb.vhdl:58:16:@20ns:(report note): Scale  : 0
./tb.vhdl:59:16:@20ns:(report note): Fract  : 00000
>% ghdl -r tb behavior --wave=test.ghw
>%
```
The report statements only show up on STD_OUTPUT when the --wave option is not given. (And it took your `repro.sh` to make it noticeable.)

tgingold, 2018-09-19:
I can reproduce the crash.

tgingold, 2018-10-26:
Note: this requires RTIs for package instances.