ghdl/ghdl #657: Missing symbols when compiling with GNAT GPL 2018 (Windows)

Paebbels, 2018-09-16:
**Description**
I downloaded the new GNAT GPL 2018 compiler from libre.adacore.com.  
When compiling GHDL 0.36-dev, I get these missing symbol errors:

**Command line call:**
```powershell
.\dist\windows\compile.ps1 -Clean -Compile
```

(*I'm using the PowerShell build script, which is calling GNAT tools like `gnatbind` or `gnatlink`.*)

**Last gnat/gcc messages and error:**
```
gnatbind -aIC:\git\GitHub\Paebbels\ghdl\dist\windows\mcode -aIC:\git\GitHub\Paebbels\ghdl\src -aIC:\git\GitHub\Paebbels\ghdl\src\ghdldrv -aIC:\git\GitHub\Paebbels\ghdl\src\psl -aIC:\git\GitHub\Paebbels\ghdl\src\grt -aIC:\git\GitHub\Paebbels\ghdl\src\ortho -aIC:\git\G
itHub\Paebbels\ghdl\src\ortho\mcode -aIC:\git\GitHub\Paebbels\ghdl\src\vhdl -aIC:\git\GitHub\Paebbels\ghdl\src\vhdl\translate -x ghdl_jit.ali
gnatlink ghdl_jit.ali -O1 -g -o ghdl.exe grt-cbinding.o clock.o grt-cvpi.o memsegs_c.o win32.o -ldbghelp
C:/GNAT/2018/lib/gcc/x86_64-pc-mingw32/7.3.1/adalib/libgnat.a(s-secsta.o):s-secsta.adb:(.text+0x40): multiple definition of `system__secondary_stack__ss_stackIP'
grt-cbinding.o:C:/git/GitHub/Paebbels/ghdl/src/grt/grt-cbinding.c:126: first defined here
C:/GNAT/2018/lib/gcc/x86_64-pc-mingw32/7.3.1/adalib/libgnat.a(s-secsta.o):s-secsta.adb:(.bss+0x8): multiple definition of `__gnat_default_ss_size'
grt-cbinding.o:C:/git/GitHub/Paebbels/ghdl/src/grt/grt-cbinding.c:120: first defined here
C:/GNAT/2018/lib/gcc/x86_64-pc-mingw32/7.3.1/adalib/libgnat.a(s-secsta.o):s-secsta.adb:(.bss+0x10): multiple definition of `__gnat_binder_ss_count'
grt-cbinding.o:C:/git/GitHub/Paebbels/ghdl/src/grt/grt-cbinding.c:119: first defined here
C:/GNAT/2018/lib/gcc/x86_64-pc-mingw32/7.3.1/adalib/libgnat.a(s-secsta.o):s-secsta.adb:(.bss+0x0): multiple definition of `__gnat_default_ss_pool'
grt-cbinding.o:C:/git/GitHub/Paebbels/ghdl/src/grt/grt-cbinding.c:121: first defined here
.\ortho_code-x86-abi.o: In function `ortho_code__x86__abi__link_intrinsics':
C:/git/GitHub/Paebbels/ghdl/src/ortho/mcode/ortho_code-x86-abi.adb:848: undefined reference to `__muldi3'
C:/git/GitHub/Paebbels/ghdl/src/ortho/mcode/ortho_code-x86-abi.adb:852: undefined reference to `__divdi3'
C:/git/GitHub/Paebbels/ghdl/src/ortho/mcode/ortho_code-x86-abi.adb:856: undefined reference to `__moddi3'
C:/git/GitHub/Paebbels/ghdl/src/ortho/mcode/ortho_code-x86-abi.adb:862: undefined reference to `__chkstk'
.\ortho_code-x86-abi.o:ortho_code-x86-abi.adb:(.rdata$.refptr.__chkstk[.refptr.__chkstk]+0x0): undefined reference to `__chkstk'
.\ortho_code-x86-abi.o:ortho_code-x86-abi.adb:(.rdata$.refptr.__moddi3[.refptr.__moddi3]+0x0): undefined reference to `__moddi3'
.\ortho_code-x86-abi.o:ortho_code-x86-abi.adb:(.rdata$.refptr.__divdi3[.refptr.__divdi3]+0x0): undefined reference to `__divdi3'
.\ortho_code-x86-abi.o:ortho_code-x86-abi.adb:(.rdata$.refptr.__muldi3[.refptr.__muldi3]+0x0): undefined reference to `__muldi3'
collect2.exe: error: ld returned 1 exit status
gnatlink: error when calling C:\GNAT\2018\bin\gcc.exe
gnatmake: *** link failed.
```

Any ideas?
Can I deliver more information for debugging?

Paebbels, 2018-09-19:
So, the switched from a 32 bit compiler to 64 bit?
It looked like if GNAT is a 64 bit tool but still produces 32 bit binaries (at least for the old revisions).

So I'll stick to GNAT GPL 2017. Any roadmap for adding the missing mcode 64 instructions?

I saw, that underneath, a MinGW64 is running. So we could also "remove" GNAT GPL 2018+ support from GHDL and advice users to use MinGW64 builds with e.g. llvm, right?

tgingold, 2018-10-25:
So yes, GNAT GPL windows before 2018 was generating 32 bit code.
GNAT GPL windows 2018 (and later ?) is generated 64 bit code, which is not yet supported by mcode.
So let's advise to use llvm back-end.

cjchin, 2019-06-01:
Just found this, was trying to figure out what was wrong. 

I think we should update this on the website, it does not say anything about using 64 bit PC when using mcode backend

tgingold, 2019-06-01:
A pull request is welcome!

byteptr, 2021-10-29:
Can anyone says me how to install CORRECTLY ghdl please? 
I found instalation instructions for windows very poor

ghdl -e --ieee=synopsys testbench_ams
C:/TDM-GCC-64/bin/../lib/gcc/x86_64-w64-mingw32/9.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lz
collect2.exe: error: ld returned 1 exit status
C:\tools\hdl\ghdl\bin\ghdl.exe: compilation error

umarcor, 2021-10-29:
@byteptr, which backend do you want to use? Which terminal/shell do you want to use?

byteptr, 2021-10-30:
Thanks for reply, good question. 
I Have three posibilities on Windows 10, first the windows command terminal, then I have also both MSYS2 and Debian through WSL. I have no special preferences, What of these tree choices is the best or easier?

umarcor, 2021-10-30:
I recommend using MSYS2. Either `pacman -S mingw-w64-i686-ghdl-mcode` (for the 32 bit mcode version) or `pacman -S mingw-w64-x86_64-ghdl-llvm` (for the 64 bit LLVM version).