ghdl/ghdl #660: Compiler bug when using GENERIC argument in range expression

hackfin, 2018-09-18:
**Description**
When using a GENERIC constant in an entity implementation, the compilation fails.

**Context**
analyze testent.vhdl
/home/strubi/src/vhdl/test-generic-parameter/testent.vhdl: In function ‘work__test_core__ARCH__behaviour__DECL_ELAB’:
/home/strubi/src/vhdl/test-generic-parameter/testent.vhdl:39:0: internal compiler error: in expand_expr_real_1, at expr.c:9843
  alias iodir_b_mux : std_logic_vector(16-1 downto 0) is
 
0x892211 expand_expr_real_1(tree_node*, rtx_def*, machine_mode, expand_modifier, rtx_def**, bool)
	/data/src/gcc-toolchain/gcc-7.2.0/gcc/expr.c:9837
0x88ec5c expand_expr_real_1(tree_node*, rtx_def*, machine_mode, expand_modifier, rtx_def**, bool)
	/data/src/gcc-toolchain/gcc-7.2.0/gcc/expr.c:10440
0x899913 store_expr_with_bounds(tree_node*, rtx_def*, int, bool, bool, tree_node*)
	/data/src/gcc-toolchain/gcc-7.2.0/gcc/expr.c:5555
0x89a829 expand_assignment(tree_node*, tree_node*, bool)
	/data/src/gcc-toolchain/gcc-7.2.0/gcc/expr.c:5321
0x7ae489 expand_gimple_stmt_1
	/data/src/gcc-toolchain/gcc-7.2.0/gcc/cfgexpand.c:3641
0x7ae489 expand_gimple_stmt
	/data/src/gcc-toolchain/gcc-7.2.0/gcc/cfgexpand.c:3737
0x7af66d expand_gimple_basic_block
	/data/src/gcc-toolchain/gcc-7.2.0/gcc/cfgexpand.c:5744
0x7b4756 execute
	/data/src/gcc-toolchain/gcc-7.2.0/gcc/cfgexpand.c:6357
Please submit a full bug report,

- OS: x86_64 GNU/Linux
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [X] Built from sources. Commit SHA: v0.35-292-gcd8247bb

**How to reproduce?**

```
#>> tb_repro.vhdl
library vunit_lib;
context vunit_lib.vunit_context;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all; -- Unsigned

entity tb_repro is
  generic ( runner_cfg : string );
end entity;

architecture tb of tb_repro is
	constant CONFIG_NUM_GPIO : natural := 2;

	signal gpio  : unsigned(CONFIG_NUM_GPIO*16-1 downto 0);
	signal gpio_i  : std_logic_vector(CONFIG_NUM_GPIO*16-1 downto 0);
	signal gpio_o  : std_logic_vector(CONFIG_NUM_GPIO*16-1 downto 0);
	signal gpio_d  : std_logic_vector(CONFIG_NUM_GPIO*16-1 downto 0);

begin
  main: process
  begin
    test_runner_setup(runner, runner_cfg);
    info("Generic argument test case");
    test_runner_cleanup(runner);
    wait;
  end process;

inst_ent: entity work.test_core
port map (
		gpio         => gpio,
		-- Inter-module connection wires for pin muxing: {
		gpio_in      => gpio_i,
		gpio_out     => gpio_o,
		gpio_dir     => gpio_d
);

end architecture;
#>> testent.vhdl

library ieee;
	use ieee.std_logic_1164.all;
	use ieee.numeric_std.all;

entity test_core is
	generic (
		CONFIG_NUM_GPIO : natural := 2;
		GP_OFFSET : natural := 16
	);
	port (
		gpio         : inout unsigned(CONFIG_NUM_GPIO*16-1 downto 0);
		-- Inter-module connection wires for pin muxing: {
		gpio_in      : out std_logic_vector(CONFIG_NUM_GPIO*16-1 downto 0);
		gpio_out     : in  std_logic_vector(CONFIG_NUM_GPIO*16-1 downto 0);
		gpio_dir     : in  std_logic_vector(CONFIG_NUM_GPIO*16-1 downto 0)

	);
end entity;

architecture behaviour of test_core is

	constant num_pins : natural := CONFIG_NUM_GPIO*16;

	-- constant GP_OFFSET : natural := 16;

	subtype PORT_A_RANGE is natural range 16-1 downto 0;
	subtype PORT_B_RANGE is integer range GP_OFFSET+15 downto GP_OFFSET;

	signal io_read : std_logic_vector(num_pins-1 downto 0);
	signal out_int : std_logic_vector(num_pins-1 downto 0);
	signal io_dir  : std_logic_vector(num_pins-1 downto 0);

	alias out_b_int_mux : std_logic_vector(16-1 downto 0) is
		out_int(31 downto 16);

	alias iodir_b_mux : std_logic_vector(16-1 downto 0) is
		io_dir(PORT_B_RANGE);

	alias gpio_b_dir : std_logic_vector(16-1 downto 0) is
		gpio_dir(PORT_B_RANGE);

	alias gpio_b_out : std_logic_vector(16-1 downto 0) is
		gpio_out(PORT_B_RANGE);

	alias io_b_read : std_logic_vector(16-1 downto 0) is
		io_read(PORT_B_RANGE);

begin

gpio_direction:
	for i in 0 to num_pins-1 generate
		gpio(i) <= out_int(i) when io_dir(i) = '1' else 'Z';
	end generate;

	io_read <= std_logic_vector(gpio);
	gpio_in <= io_read;

mux_port_b:
	process (gpio_b_out, gpio_b_dir)
	begin
		-- default:
		out_b_int_mux <= gpio_b_out;
		iodir_b_mux   <= gpio_b_dir;

	end process;
end architecture;
#>> run.py
from os.path import join, dirname
from vunit import VUnit

vu = VUnit.from_argv()
# vu.add_library("lib").add_source_files(join(dirname(__file__), "*.vhdl"))
l = vu.add_library("lib")
l.add_source_file("testent.vhdl")
l.add_source_file("tb_repro.vhdl")
vu.main()

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ext:vunit python run.py -v
#>> end
```

- [X] Add `GHDL Bug occurred` log block
- [X] Add a MWE
- [X] Try the latest version

tgingold, 2018-09-18:
I think this is the same issue as #649, and therefore is now fixed.

hackfin, 2018-09-18:
That's one of the quickes fixes ever found/closed/verified working. Thanks for the good job!