ghdl/ghdl #661: assertion warning numeric_std: null detected, returning 0

Ravenwater, 2018-09-20:
**Description**
assertion warning

**Expected behaviour**
Can't locate what is triggering this assertion

A test is generating this set of assertion warnings, but it doesn't provide any context on what is triggering the assertion:
numeric_std-body.v08:3027:7:@10ns:(assertion warning): NUMERIC_STD.TO_INTEGER: null detected, returning 0
numeric_std-body.v08:3027:7:@15ns:(assertion warning): NUMERIC_STD.TO_INTEGER: null detected, returning 0
numeric_std-body.v08:3027:7:@20ns:(assertion warning): NUMERIC_STD.TO_INTEGER: null detected, returning 0
numeric_std-body.v08:3027:7:@25ns:(assertion warning): NUMERIC_STD.TO_INTEGER: null detected, returning 0
numeric_std-body.v08:3027:7:@30ns:(assertion warning): NUMERIC_STD.TO_INTEGER: null detected, returning 0
numeric_std-body.v08:3027:7:@35ns:(assertion warning): NUMERIC_STD.TO_INTEGER: null detected, returning 0
numeric_std-body.v08:3027:7:@40ns:(assertion warning): NUMERIC_STD.TO_INTEGER: null detected, returning 0
numeric_std-body.v08:3027:7:@45ns:(assertion warning): NUMERIC_STD.TO_INTEGER: null detected, returning 0
numeric_std-body.v08:3027:7:@50ns:(assertion warning): NUMERIC_STD.TO_INTEGER: null detected, returning 0
numeric_std-body.v08:3027:7:@55ns:(assertion warning): NUMERIC_STD.TO_INTEGER: null detected, returning 0

How do I debug this?

1138-4EB, 2018-09-20:
Hi @Ravenwater! That is a quite common warning, when you use `std_logic_vector` signals as `unsigned` or `signed` to compare/assign integer values. There are some states that `std_logic` can have, but which cannot be represented as integers, e.g. 'U' or 'X'.

I think that you might avoid it if you initialize all the signals when you declare them, and ensure that only valid states occur (`0` or `1`).

Alternatively, you can use option [`--ieee-asserts<=POLICY>`](https://ghdl.readthedocs.io/en/latest/using/Simulation.html?highlight=warning#cmdoption-ghdl-ieee-asserts) in GHDL.

faf5678, 2018-09-20:
this particular assert is about the size of the unsigned argument to to_integer being of size 0 bits.

also, it would be much easier to ensure that the argument to to_integer is proper if there was a stack backtrace.

what you can do now @Ravenwater is to manually check for every call to to_integer whether the argument size is 0.