ghdl/ghdl #662: Support of PSL functions

tmeissner, 2018-09-25:
**Is your feature request related to a problem? Please describe.**
Maybe you have following code:

```vhdl
Pipeline : process (clk) is
begin
  if rising_edge(clk) then
    if Wen_i = '1' then
      Data_o <= Data_i;
    end if;
    Wen_o <= Wen_i;
  end if;
end process Pipeline;
```

If you want to check that Data_o is valid in a write cycle, you need the content of Data_i in the cycle before, for example by using the PSL function prev():

```vhdl
-- psl assert always Wen_o -> Data_o = Data_i;
```

**Describe the solution you'd like**
There are various functions defined in the PSL specification which GHDL doesn't implement. For example the prev() function, which returns the value of a signal in the previous cycle (useful in the above example). The stable() function is also very useful to check if signals are stable during a range of clock cycles.

A list of PSL functions in order of usefulness:

Would be really useful:
* prev()
* stable()

Not so important, can be relatively easy implemented is VHDL functions instead:
* rose()
* fell()
* isunknown()
* countones()
* onehot()
* onehot0()

Don't know if these functions are useful or valid in the simple subset:
* nondet()
* nondet_vector()
* ended()
* next()

Not so important because it can use PSL endpoints instead, see #45:
* ended()

**Describe alternatives you've considered**
Some of these function could be emulated by using (a lot of) VHDL code which implement their behaviour.

diogratia, 2018-09-26:
IEEE Std 1076-2008 1.3.5 Incorporation of Property Specification Language:

>VHDL incorporates the simple subset of the Property Specification Language (PSL) as an embedded language for formal specification of the behavior of a VHDL description. PSL is defined by IEEE Std 1850-2005. All PSL constructs that appear in a VHDL description shall conform to the VHDL flavor of PSL. Within this standard, reference is made to syntactic rules of PSL. Each such reference has the italicized prefix PSL_ and corresponds to the syntax rule in IEEE Std 1850-2005 with the same name but without the prefix.

(Solely to point out there isn't a revision of the VHDL standard required to conform to IEEE Std 1850-2010. Also as a normative reference  in 2. Normative references.)

All the listed functions are found in IEEE Std 1850-2005, see 5.2.3 Built-in functions.

tmeissner, 2018-09-26:
I only have the 2010 standard, that’s the reason why I referenced it 😉

And, correct me if I’m wrong, VHDL-2018 will include PSL 2010 instead of 2005. But I know of none vendor which implements 2010. Mentor for example don’t even implements 2005 completely.

tgingold, 2018-09-27:
I understand that prev() would be very useful, but could stable() be replaced by 'stable ?

tmeissner, 2018-09-27:
'stable works as intended. For example:

```vhdl
-- psl WISHBONE_DOUT_STABLE : assert always
--   WbWe_o -> next WbDat_o'stable until (WbErr_i or WbAck_i);
```

However, I've stumbled over another thing. The until operator holds true till the cycle direct before a given requirement. The until_ operator includes the cycle where the requirement holds. However when I use the until_ operator in the property above, I get an error during analyzing (WbErr_i and WbAck_i are std_logic):

```vhdl
-- psl WISHBONE_DOUT_STABLE : assert always
--   WbWe_o -> next WbDat_o'stable until_ (WbErr_i or WbAck_i);
```

```
error: both operands of an overlapping 'until*' operator are boolean
```

Is the return value (WbErr_i or WbAck_i) interpreted as boolean? In VHDL this would be an std_logic, wouldn't it? To make it more clear, when I only use WbAck_i:

```vhdl
-- psl WISHBONE_DOUT_STABLE : assert always
--   WbWe_o -> next WbDat_o'stable until_ WbAck_i;
```

I get the same error.

pepijndevos, 2019-08-14:
This code gives me `synth_declaration: cannot handle IIR_KIND_SIGNAL_ATTRIBUTE_DECLARATION (alu.vhd:80:27)`. Has this regressed or am I doing something wrong? I'm pretty much doing the exact thing @tmeissner was doing: `-- psl assert always rst_n -> next opcode'stable until (rst_n);`

tgingold, 2019-08-14:
'stable is supported by simulation but not by synthesis.

tgingold, 2020-06-02:
`prev` is now supported in synthesis.  That's the first step.

tmeissner, 2020-06-02:
Oh, cool. I could create some test cases for the other function, if you like. I add create examples of PSL functions  to my [psl_with_ghdl](https://github.com/tmeissner/psl_with_ghdl) anyway

tmeissner, 2020-06-02:
I have added an example for the PSL `prev()` function to my repo: [PSL prev() example](https://github.com/tmeissner/psl_with_ghdl/blob/master/src/psl_prev.vhd). What is left with `prev()`: allowing vectors as parameter.

tgingold, 2020-06-02:
Do you have a testcase for prev() with vectors ?  psl_prev.vhdl can be synthesized.

tmeissner, 2020-06-02:
Yes, I have added some checks with vectors: [prev() example](https://github.com/tmeissner/psl_with_ghdl/blob/d904d45c9b27072ef0c0bce5b3e5bb4d743582da/src/psl_prev.vhd#L73). They all work successfully, it was false alarm :smile:

tmeissner, 2020-06-02:
I've added examples for PSL functions `rose()`, `fell()` and `stable()` in my psl_with_ghdl repo:

- [rose() example](https://github.com/tmeissner/psl_with_ghdl/blob/master/src/psl_rose.vhd)
- [fell() example](https://github.com/tmeissner/psl_with_ghdl/blob/master/src/psl_fell.vhd)
- [stable() example](https://github.com/tmeissner/psl_with_ghdl/blob/master/src/psl_stable.vhd)

Fell free to use them as test cases 😃