ghdl/ghdl #663: eb9974cf1578e1bd4f4c167601fa767398db3e80 broke VUnit regression

kraigher, 2018-09-26:
**Description**
VUnit regression tests where broken by https://github.com/ghdl/ghdl/commit/eb9974cf1578e1bd4f4c167601fa767398db3e80

I tried and it worked in the parent commit.

```
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 4.8
Target: x86_64-linux-gnu
In directory: /home/travis/build/VUnit/vunit/
Command line:
/home/travis/build/VUnit/vunit/install-ghdl-llvm/bin/ghdl1-llvm --workdir=/home/travis/build/VUnit/vunit/vunit/test/acceptance/external_run_out/ghdl/libraries/lib --work=lib --std=02 -P/home/travis/build/VUnit/vunit/vunit/test/acceptance/external_run_out/ghdl/libraries/vunit_lib/ -P/home/travis/build/VUnit/vunit/vunit/test/acceptance/external_run_out/ghdl/libraries/lib/ -P/home/travis/build/VUnit/vunit/vunit/test/acceptance/external_run_out/ghdl/libraries/logging_tb_lib/ -P/home/travis/build/VUnit/vunit/install-ghdl-llvm/lib/ghdl/ieee/v93/ -P/home/travis/build/VUnit/vunit/install-ghdl-llvm/lib/ghdl/ -c -o /home/travis/build/VUnit/vunit/vunit/test/acceptance/external_run_out/ghdl/libraries/lib/tb_check_stable.o /home/travis/build/VUnit/vunit/.tox/py36-acceptance-ghdl/lib/python3.6/site-packages/vunit/vhdl/check/test/tb_check_stable.vhd
Exception CONSTRAINT_ERROR raised
Exception information:
Exception name: CONSTRAINT_ERROR
Message: trans-chap3.adb:706 access check failed
Call stack traceback locations:
0x5d30e7 0x5e21e4 0x5c6f66 0x5cb2ac 0x5a017a 0x5f61cf 0x582ac4 0x5fb28c 0x5fbaae 0x478092 0x43217c 0x7f66414bcf43 0x431510 0xfffffffffffffffe
******************************************************************
```

**How to reproduce?**
Checkout VUnit master and run this:
```
python vunit/vhdl/check/run.py
```

kraigher, 2018-10-24:
@tgingold I have made a smaller reproducer for your convenience.
Seem the problem is related to aliases.

**ent.vhd** 
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity ent is
end entity ent;

architecture a of ent is
  signal clk : std_logic := '0';

  signal check_stable_in_1 : std_logic_vector(1 to 5) := "00000";
  alias check_stable_start_event_1  : std_logic is check_stable_in_1(1);
  alias check_stable_end_event_1    : std_logic is check_stable_in_1(2);
  alias check_stable_expr_1         : std_logic_vector(2 downto 0) is check_stable_in_1(3 to 5);

  signal check_stable_en_1 : std_logic;

  signal en, start_event, end_event, expr : std_logic := '1';

  procedure check_stable(
    signal clock               : in std_logic;
    signal en                  : in std_logic;
    signal start_event         : in std_logic;
    signal end_event           : in std_logic;
    signal expr                : in std_logic_vector) is
  begin
    wait until (falling_edge(clock) or rising_edge(clock)) and (to_x01(en) = '1');
  end;

begin
  clock : process is
  begin
    clk <= '1', '0' after 5 ns;
  end process clock;

  check_stable_1 : check_stable(clk,
                                check_stable_en_1,
                                check_stable_start_event_1,
                                check_stable_end_event_1,
                                check_stable_expr_1);

end architecture;

```

```
set -ex
ghdl --version
ghdl -a ent.vhd
ghdl --elab-run ent
```

tgingold, 2018-10-24:
Thanks, that's *very* useful.  I thought the issue was with vhdl 2008, but not.

commit 47a0b02c (2018-10-25): trans-chap9: add a guard.

Fix #663
closed 2018-10-25 by tgingold