ghdl/ghdl #666: Segmentation fault for long string

slaweksiluk, 2018-10-09:
**Description**
String with length of 50331648 (48x1024x1024) causes Segmentation fault.

**Expected behaviour**
No Segmentation fault.

**Context**
Segmentation fault (core dumped)
$ghdl -v
GHDL 0.36-dev (v0.35-312-g1cbe9b82) [Dunoon edition]
 Compiled with GNAT Version: 7.3.1 20180712 (Red Hat 7.3.1-6)
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

- OS:
Fedora 27
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: 1cbe9b82

**Additional context**
Bug occured while tried to use Vunit library for encoding custom record types to string to push it to queue. Such a big string might means Vunit internal error but still, ghdl sholud not segfault.

**How to reproduce?**
```
#>> tb.vhd
entity tb is
end entity;
architecture bench of tb is
	constant kill_size : positive := 50331648;
	signal s : string(1 to kill_size);

	function fun return string is
		variable v : string(1 to kill_size);
	begin
		return "hello";
	end;
begin
	proc: process
		-- Segmentation fault occurs for variable too, but much faster
--		variable s : string(1 to kill_size);
	begin
--		s := fun;
		s <= fun;
		report "pass" severity failure;
	end process;
end bench;

#>> sim.sh
ghdl -a tb.vhd
ghdl --elab-run tb

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

tgingold, 2018-10-19:
I am not sure if there is really a bug.  It's like allocating a huge local variable in C.

Paebbels, 2018-10-19:
@tgingold Can you profile how many memory is allocated? @slaweksiluk reported 50 MB, right? That should be a problem, whereas 500 MB would be different. Otoh, GHDL 64 bit should be able to handle and allocated even bigger chunks. (Regardless of usefulness and simulation speed.)

C and malloc would cause a "could not allocate" / "out of memory" error, but no segmentation fault. A segmentation fault is related to an access error in existing memory. I don't know the pendant in Ada for `malloc`, but if it fails because of not enough memory in application memory space or because the object is to big, it should report when you allocate that peace of memory, right?

slaweksiluk, 2018-10-19:
Does ghdl use malloc to allocate constants instead of statically allocated object? I don't think 48MB should be a problem for a malloc. If static allocation is used I think some compile error should be raised, as run time segmentation fault is not good hint for what just happened.

Paebbels, 2018-10-19:
That's a good question for @tgingold, if the object is allocated on the stack, which has other object size limitations.

tgingold, 2018-10-25:
The crash occurs in function fun, which allocates v on the stack.
By checking stack allocation, the message could be improved, but it's heavy.
Not worth yet IMHO.

closed 2018-10-25 by tgingold

tgingold, 2018-10-25:
No, the allocatable size is for all the frames (so all the calls).

I am not sure we need to give a workaround, given it was a mistake.  Using ulimit it is also possible to increase the size of the stack.