ghdl/ghdl #667: Compiling VUnit context produces CBZ WZR / CBNZ xZR branches on ARM platforms

1138-4EB, 2018-10-09:
This is neither a bug nor a feature request. It is just an interesting issue that came out when using GHDL with tools at machine level. The context is the following:

- GHDL is built natively in an ARM target. LLVM backend with `-fPIC` enabled; see *Context* of #640.
- VUnit context is loaded in a testbench, and this is compiled with GHDL.
- The disassembled (gdb) binary (produced after elaboration, `ghdl -e`) contains several branch instructions such as `CBZ WZR <address>` or `CBNZ xZR <address>`, where `CBZ`/`CBNZ` mean *compare and branch if (not) zero*; and `WZR`/`xZR` are hardcoded zero registers (32/64 bits respectively).

Instructions such as `CBZ WZR <address>` don't make sense. Comparing `WZR` with zero will always be true. So, it should have been compiled to `B <address>` (a direct branch, not a conditional one). By the same token, `CBNZ xZR <address>` should be a `nop` or not exists at all.

The resulting code (although ugly) works ok. I am just curious about why such strange code is generated from VUnit and not from other VHDL sources. However, I don't know how to further debug this:

- Is it because how VUnit libraries are coded?
- Is it because how GHDL manages/manipulates sources during elaboration?
- Does it have nothing to do with either VUnit or GHDL, and is this done by LLVM?

I'd appreciate any hint, even if it involves not being worth to dig further.

tgingold, 2018-10-10:
The number of LLVM optimisation passes enabled in ghdl is very limited.  That's certainly something to improve.
Once enabled, I suppose these instructions shouldn't appear

1138-4EB, 2018-10-10:
Thanks @tgingold. Is that something I can tinker with? Or is it an advanced task?

tgingold, 2018-10-10:
No, it is not an advanced task.

LLVM has a long list of optimization passes.  You need to find the most interesting ones and the best order.

There is a couple of passes already added in src/ortho/llvm-nodebug/ortho_code_main.adb (See code about PassManager).

Optimizations are enabled with -O.

tgingold, 2018-12-12:
I close this issue as it was answered.  If you need more help, do not hesitate to reopen it.

closed 2018-12-12 by tgingold