ghdl/ghdl #668: error: wave.create_type

slaweksiluk, 2018-10-10:
**Description**
Error while trying to dump subtype of unconstrained record.

**Expected behaviour**
Dump ghw

**Context**
ave-create-error2> ./run.sh 
without ghw
a.vhd:45:16:@100ns:(report failure): pass
./wb_demux_tb:error: report failed
./wb_demux_tb:error: simulation failed
with ghw
./wb_demux_tb:internal error: wave.create_type

- OS: Fedora 27
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: 1cbe9b8

**Additional context**
Similar to https://github.com/ghdl/ghdl/issues/561

**How to reproduce?**
```vhdl
#>> ent.vhd
library ieee;
use ieee.std_logic_1164.all;

package wishbone_pkg is
--./wb_demux_tb:internal error: wave.create_type
	type t_wishbone_master_out is record
		dat : std_logic_vector;
	end record;
	subtype t_wishbone_slave_in is t_wishbone_master_out;
end wishbone_pkg;

library IEEE;
use IEEE.STD_LOGIC_1164.ALL;

library work;
use work.wishbone_pkg.all;

entity wb_demux is
    Port (
    	wbs_i	: in t_wishbone_slave_in
    );
end wb_demux;
architecture full_regs of wb_demux is
begin
end full_regs;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;
use work.wishbone_pkg.all;

entity wb_demux_tb is
end entity;
architecture bench of wb_demux_tb is
	signal wbs_i : t_wishbone_slave_in(
		dat(32-1 downto 0)
	);
begin
	stimulus : process
	begin
		wbs_i.dat <= x"deadbeef";
		wait for 100 ns;
		report "pass" severity failure;
	end process;

	dut : entity work.wb_demux
	port map (
		wbs_i => wbs_i
	);
end architecture;

#>> sim.sh
ghdl -a --std=08 a.vhd
ghdl -e --std=08 wb_demux_tb
echo "without ghw"
./wb_demux_tb
echo "with ghw"
./wb_demux_tb --wave=w.ghw

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

tgingold, 2018-10-17:
I can reproduce.  It will certainly need an update of GtkWave too.

closed 2018-10-21 by tgingold

1138-4EB, 2018-10-21:
@tgingold, is the update already pushed to GtkWave or do we need to wait for it?

tgingold, 2018-10-21:
At the end, there was no change in GtkWave.

commit ffa1a498 (2018-10-21): grt rtis/wave: handle unbounded record subtypes.

Fix #668