ghdl/ghdl #672: --time-resolution= vs run time

wdgConsulting, 2018-10-15:
I have medium size project with absurd simulation run time compared with XSIM run time.
Trying to reproduce the problem with simple sources I found:

--stop-time=1us --time-resolution auto => run time 1s
--stop-time=1us --time-resolution 1us => run time 1s
--stop-time=1us --time-resolution 1ns => run time 95s

[SQRT.zip](https://github.com/ghdl/ghdl/files/2480117/SQRT.zip)

wdgConsulting, 2018-10-15:
I'm using 
GHDL 0.36-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 7.3.0
 mcode code generator

Paebbels, 2018-10-15:
I run the provided testbench on my machine and I get these results (all run with `stop-time=1us`):

**--time-resolution=fs**  
runtime of 0 seconds

**--time-resolution=ps**  
runtime of 0 seconds

**--time-resolution=ns**  
runtime of 20 seconds

**--time-resolution=us**  
GHDL reports 4 errors:
> SQRTb.vhd:26:28: physical unit "ns" is below the time resolution  
> SQRTb.vhd:49:16: physical unit "ns" is below the time resolution  
> SQRTb.vhd:50:16: physical unit "ns" is below the time resolution  
> SQRTb.vhd:52:16: physical unit "ns" is below the time resolution

**--time-resolution=auto**  
GHDL reports:
> Time resolution is 1 fs

Runtime of 0 seconds.

------------------

@tgingold 
* option value `auto` is not documented in `ghdl --help -r`  
  >  --time-resolution=UNIT   Set the resolution of type time  
  >  UNIT can be fs, ps, ns, us, ms, sec, min or hr
* option `--time-resolution` is undocumented in ReadTheDocs, or have I missed it somewhere?
* according to VHDL-2008, a time value below the time resolution limit is allowed, as far as the value itself is zero. Thus the error message should be improved to something like: `The time 50 ns is below the time resolution of 1 us`.
   
   > By default, the primary unit of type TIME (1 fs) is the resolution limit for type TIME. Any TIME value
whose absolute value is smaller than this limit **is truncated to zero (0) time units.** An implementation may allow a given elaboration of a model (see Clause 14) to select a secondary unit of type TIME as the resolution limit. Furthermore, an implementation may restrict the precision of the representation of values of type TIME and the results of expressions of type TIME, provided that values as small as the resolution limit are representable within those restrictions. **It is an error if a given unit of type TIME** appears anywhere within the design hierarchy defining a model to be elaborated, **and if the position number of that unit is less than that of the secondary unit selected as the resolution limit for type TIME** during the elaboration of the model, **unless that unit is part of a physical literal whose abstract literal is either the integer value zero or the floating-point value zero.**

   It ready a bit contradictory, because on the one hand, all value below the resolution limit are set to zero, on the other hand it's an error.

1138-4EB, 2018-10-15:
@Paebbels, no, it is not documented in RTD. It was added 'just' a couple of months ago, and it is for mcode backend only AFAIK. See https://github.com/ghdl/ghdl/issues/613#issuecomment-411631499

About performance itself, since GHDL is single threaded and CPU bound, it'd be interesting to know the clock frequency of the machines you are both using. That might explain the 5X difference between result reported by @wdgConsulting and @Paebbels.

Last, mcode backend is expected to analyze faster but run slower than LLVM/GCC: https://github.com/ghdl/ghdl#building-ghdl Therefore, it might be interesting to check execution time with a different backend.

wdgConsulting, 2018-10-15:
@Paebbels 
Your are right, sorry, my typo. 
"--stop-time=1us --time-resolution 1us => run time 1s"
should be
"--stop-time=1us --time-resolution 1ps => run time 1s" 

If I reduce the resolution time from ps to ns I will speed up my simulation; so,

--time-resolution=ps
runtime of 0 seconds

--time-resolution=ns
runtime of 20 seconds

Is an unexpected behaviour. Correct ?

With full VHDL sources, I find a similar issue using --time-resolution=ps. Can't test ns due Xilinx libs.

wdgConsulting, 2018-10-16:
Running additional tests 
Same VHDL sources script changes only

[image]

From wave forms, apparently time references are broken 
.

tgingold, 2018-10-18:
I can reproduce the issue.

Which format are you using for the traces ?  vcd or ghw ?

tgingold, 2018-10-18:
I think that the stop-time value is not scaled according to time-resolution.

wdgConsulting, 2018-10-18:
Posted waveform was ghw;

tgingold, 2018-10-19:
Ok, fixing the output requires a change in GtkWave.  Not right now.

wdgConsulting, 2018-10-19:
Ok, but the original question was about time-resolution vs run time 
 
--stop-time=1us --time-resolution auto => run time 1s
--stop-time=1us --time-resolution 1ps => run time 1s
--stop-time=1us --time-resolution 1ns => run time 95s

Is this the expected behaviour ?

tgingold, 2018-10-19:
No, this is not expected.  But I have a fix for that.

wdgConsulting, 2018-10-19:
Excellent. Have you an estimated release date ?

Paebbels, 2018-10-20:
@tgingold If this issue is related to the stop time not being scaled correctly, is this then a bug instead of a performance questions, because the simulations results are not correct?

tgingold, 2018-10-21:
I have just pushed the fix.

commit 1803f976 (2018-10-21): Pass time-resolution to grt.

Use time-resolution for Put_Time.
Use time-resolution for --stop-time (Parse_Time).
Use time-resolution as VpiTimePrecision and for vcd.
TODO: ghw (but needs to update ghwlib and gtkwave)

Fix #672
closed 2018-10-21 by tgingold