ghdl/ghdl #673: Build from source fails during make install using GNAT 2018

byzantic, 2018-10-16:
Build from source fails during `make install` using GNAT 2018

Last output from make:

```
make[1]: Entering directory '/home/rob/sourcecode/ghdl/build'
gnatmake -o ghdl1-llvm -aI../src/ortho/llvm4-nodebug -aI../src/ortho \
-aI../src -aI../src/vhdl -aI../src/grt -aI../src/psl -aI../src/vhdl/translate -aI../src/ghdldrv -aI../src/ortho -aI../src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe ortho_code_main -bargs -E \
-largs llvm-cbindings.o --LINK=clang++ \
 `llvm-config --ldflags --libs --system-libs`
/bin/sh: 1: gnatmake: not found
../src/ortho/llvm4-nodebug/Makefile:12: recipe for target 'ghdl1-llvm' failed
make[1]: *** [ghdl1-llvm] Error 127
make[1]: Leaving directory '/home/rob/sourcecode/ghdl/build'
Makefile:305: recipe for target 'ghdl1-llvm' failed
make: *** [ghdl1-llvm] Error 2
```
Ubuntu 18.04, using LLVM backend with LLVM version LLVM version 6.0.0, Ada Compiler is GNAT 2018 Community

Source commit is 43c96a71ddfb17443cc1bbf40833c821e70d07b5

My problems started after I updated to Ada 2018 (I use it for other, Ada based projects, otherwise I would not have updated). I found the issue about using the -05 flag, and tried to fix this manually in the Makefiles, but with no success (actually I think I hit the same problem above). I then updated the source to the latest commit and tried again. `make` succeeds OK, but `make install` fails with the above problem.

From the log, it looks as though `gnatmake` is not being found, but it seems accessible both from bash and sh, so I can't quite work out what is going on. Any ideas?

byzantic, 2018-10-16:
Hmm. A bit weird. If I execute that command line in the build dir, it just comes back: `ghdl1-llvm is up to date`. And indeed, there seems to be an executable ghdl1-llvm in th build directory that is fresh.

On further investigation, I think that the line that fails is triggered from this target rule in Makefile (line 318), which causes a rebuild of `ghdl1-llvm$(EXEEXT)`
```make
install.llvm.program: install.dirs ghdl1-llvm$(EXEEXT) ghdl_llvm$(EXEEXT)
	$(INSTALL_PROGRAM) ghdl_llvm$(EXEEXT) $(DESTDIR)$(bindir)/ghdl$(EXEEXT)
	$(INSTALL_PROGRAM) ghdl1-llvm$(EXEEXT) $(DESTDIR)$(bindir)/ghdl1-llvm$(EXEEXT)
```
and `ghdl1-llvm$(EXEEXT)` is built using the following rule on line 304
```make
ghdl1-llvm$(EXEEXT): version.ads force
	$(MAKE) -f $(srcdir)/src/ortho/$(llvm_be)/Makefile \
	 ortho_srcdir=$(srcdir)/src/ortho ortho_exec=$@ \
	 GNATFLAGS="$(GHDL_LLVM_INCFLAGS) $(GNATFLAGS)" LDFLAGS="$(LDFLAGS)" \
	 LLVM_CONFIG="$(LLVM_CONFIG)" CXX="$(CXX)" GNATMAKE="$(GNATMAKE)" \
	 all
```
which does indeed invoke the Makefile as mentioned in the original make failure.

So maybe it's some sort of timestamping problem; `make install` tries to rebuild ghdl1-llvm when it shouldn't ..

tgingold, 2018-10-17:
Yes, 'make install' tries to rebuild the binary.
Only gnatmake knows about the Ada dependencies, so make has to call gnatmake to check.

I could remove the dependencies of 'install', and therefore 'make install' will install what is currently built, but this is not the usually expected behaviour.

For me this is just a minor, mostly unfixable annoyance.

byzantic, 2018-10-17:
Hi .. I didn't actually mean that it shouldn't do it as part of a normal `make install`, I just wondered whether there was some sort of race that was causing problems.

At the moment, I can't complete the install, and if the executables are actually not correct, then simply performing the install by hand is not the correct thing to do ..

byzantic, 2018-10-17:
*sigh*
OK, I think it's a PATH issue. The recursive call to make in the ../src/ortho/llvm4-nodebug directory, seems to use /bin/sh rather than /bin/bash, but also seems to bypass the PATH set up in ~/.profile. In this case, GNAT is installed to a directory under /opt, so it does not appear in /bin /usr/bin .. etc., and hence gnatmake is not found.

At the moment I can get round it by hardwiring the GNATMAKE path in the top-level Makefile.in, but I will try to find a better fix, and probably close the issue soon.

tgingold, 2018-10-17:
But PATH should be inherited, no ?

byzantic, 2018-10-17:
I'm trying to work out what actually happens, but there are a few subtleties (= gotchas ) with make, so it may not be obvious. 

I suspect it may be some peculiarity of my setup, since you have not had any problems.. though I try hard _not_ to have anything special going on.

byzantic, 2018-10-18:
Mystery solved. I was actually running 
`>sudo make install`

But this _doesn't_ use the current PATH, it uses a minimum PATH, and so the GNAT tools are not seen.I think the path policy for sudo must have changed in Ubuntu 18.04, as I built 0.35 happily before ..

I can build by manipulating the command line for building:

`>sudo PATH=/opt/GNAT/2018/bin:$PATH make install`

Closing this, as it's not a bug in ghdl.

closed 2018-10-18 by byzantic