ghdl/ghdl #674: NULL access dereferenced

slaweksiluk, 2018-10-17:
**Description**
NULL access dereferenced

**Context**
```
       185000000 fs - tb                           -    INFO - wait on data
../../src/ieee2008/numeric_std-body.vhdl:1774:7:@215ns:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
/home/slawek/Dokumenty/hdl/test/dma_controller_tb.vhd:204:24:@255ns:(report note): axis_m_valid 1
/home/slawek/Dokumenty/hdl/test/dma_controller_tb.vhd:205:24:@255ns:(report note): axis_m_data0 0
/home/slawek/Dokumenty/hdl/test/dma_controller_tb.vhd:206:24:@255ns:(report note): axis_m_data1 
/home/slawek/Dokumenty/hdl/test/dma_controller_tb.vhd:207:24:@255ns:(report note): axis_m_data30 0
/home/slawek/Dokumenty/hdl/test/vunit_out/test_output/lib.dma_controller_tb.mm2s_8eb32eb7d236076519a7ba6d8afe440dcdaee18b/ghdl/dma_controller_tb-bench:error: NULL access dereferenced
/home/slawek/Dokumenty/hdl/test/vunit_out/test_output/lib.dma_controller_tb.mm2s_8eb32eb7d236076519a7ba6d8afe440dcdaee18b/ghdl/dma_controller_tb-bench:error: simulation failed
```
```
test> ghdl -v
GHDL 0.36-dev (v0.35-312-g1cbe9b8) [Dunoon edition]
 Compiled with GNAT Version: GPL 2017 (20170515-63)
 GCC back-end code generator
Written by Tristan Gingold.
```

- OS: Ubuntu 14
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: 1cbe9b8

**How to reproduce?**
Need vunit
```
> PYTHONPATH=~/git/github/vunit python run.py
```
**Files**

[ghdl-crash.zip](https://github.com/ghdl/ghdl/files/2489098/ghdl-crash.zip)

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

tgingold, 2018-11-11:
I do not have anymore a NULL access error, but a bound check failure.

So it seems that a check was missing.

slaweksiluk, 2018-11-12:
Yes, this issue is duplicate of #687

closed 2018-11-12 by slaweksiluk