ghdl/ghdl #675: Build with mcode backend failing on some platforms

1138-4EB, 2018-10-21:
As reported a couple of months ago (see https://github.com/ghdl/ghdl/issues/584#issuecomment-412486087), the testsuite has been failing on Debian Buster and Ubuntu 18, with mcode backend, because of **gna issue584**. The same code base passed successfully on any other platform (Debian Stretch; Ubuntu 14 and 16; Fedora 26, 27 and 28). E.g.:

- Run two months ago: https://travis-ci.com/ghdl/docker/builds/81616596
- Run yesterday: https://travis-ci.com/ghdl/docker/builds/88614822

These were not critical, because we run them in the *extended platform list* at [travis-ci.com/ghdl/docker](https://travis-ci.com/ghdl/docker/builds).

However, one of the seven commits pushed today (which where commited on Oct 16 or Oct 21, https://github.com/ghdl/ghdl/commits/master) makes mcode fail when running the testsuite on Debian Stretch or Fedora 28. So, the main project is failing: [travis-ci.org/ghdl/ghdl](https://travis-ci.org/ghdl/ghdl/builds). Precisely, **gna bug077** produces a `NULL access dereferenced` error: https://travis-ci.org/ghdl/ghdl/jobs/444320395#L1372. As a result, 6/8 platforms in the extended list are failing: https://travis-ci.com/ghdl/docker/builds/88645222

Note that this is only related to mcode backend. LLVM and GCC pass successfully.

tgingold, 2018-10-21:
Yes, I plan to investigate soon.  That doesn't appear on my computer...

tgingold, 2018-10-24:
Now fixed.  Thanks!

closed 2018-10-24 by tgingold