ghdl/ghdl #676: VHDL-2008 assignment to aggregates crashes analysis

xnorme, 2018-10-23:
**Description**
VHDL-2008 allows aggregates to be target of assignments. See e.g. [here](https://www.doulos.com/knowhow/vhdl_designers_guide/vhdl_2008/vhdl_200x_ease/#aggregates) or the LRM. GHDL crashes when it sees such a statement, despite the std=08 flag.  

**Expected behaviour**
It would be great if this feature is actually implemented. Otherwise an error message should be printed describing the limitation and pointing the user to the troubling statement's line number and filename.

**Context**
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-326-g59fe6113) [Dunoon edition]
Compiled with GNAT Version: 8.1.0
Target: x86_64-apple-darwin15
In directory: /Users/kamyar/eclipse-workspace/ECE545/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -o adder.o adder.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : sem_expr.adb:3506
Load address: 0x100217000
Call stack traceback locations:
0x1002d3425 0x1002d43f3 0x1002d4a6a 0x1002d6a5b 0x10029eb2f 0x10029f5e8 0x1002a2a6d 0x1002a2e7b 0x1002d7d34 0x1002ded00 0x10029567a 0x10044e533 0x10044f13f 0x1004509ee 0x10021a832
******************************************************************

- OS: macOS Mojave, llvm 6 and 7
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [ * ] Built from sources. Commit SHA: 459ffbfa71366beb87436f02352e33d31c700aab

**Additional context**

**How to reproduce?**

```
#>> adder.vhd
library ieee ;
use ieee.std_logic_1164.all;
use ieee.numeric_std_unsigned.all;

entity Adder is
	
generic(
    N : positive := 4
    );
    
port(
    A    : in  std_logic_vector(N-1 downto 0);
    B    : in  std_logic_vector(N-1 downto 0);
    Cin  : in  std_logic;
    Sum  : out std_logic_vector(N-1 downto 0);
    Cout : out std_logic
    );
    
end Adder;

architecture RTL of Adder is
	signal cout_sum: std_logic_vector(Sum'length downto 0);
begin

--	This works fine:
--  cout_sum <= ("0" & A) + B + Cin;
--	Cout <= cout_sum(Sum'length);
--	Sum <= cout_sum(Sum'length-1 downto 0);

-- This crashes GHDL:
	(Cout, Sum) <= ("0" & A) + B + Cin;

end RTL; 

#>> sim.sh
ghdl -a --std=08  adder.vhdl

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Files**

**Checklist**
Before submitting your issue, please review the following checklist:

- [ * ] Add `GHDL Bug occurred` log block
- [ * ] Add a MWE
- [ * ] Try the latest version

Paebbels, 2018-10-23:
Assigning to aggregates is part of VHDL-93, but assigning to aggregates where elements and arrays are mixed is new in VHDL-2008.

This is related to #643.

xnorme, 2018-10-23:
Thank you so much for the quick response and also the helpful clarification. Yes you are absolutely right, and indeed having aggregates of arrays or elements alone seems to work just fine. I'm editing the title to reflect your clarification.

Paebbels, 2018-10-28:
This is a duplicate of #642.

Paebbels, 2018-11-17:
@tgingold as #642 is closed, is this also fixed?

tgingold, 2018-11-17:
Not yet :-(

Paebbels, 2018-12-04:
Last commit added a test case for this issue, but it did not close.

It this by intention not closed yet?

tgingold, 2018-12-04:
Yes, I have to test it more.

Paebbels, 2018-12-29:
Any updates for this issue?

tgingold, 2018-12-29:
I close it.

closed 2018-12-29 by tgingold