ghdl/ghdl #678: PowerShell installation fail on Windows 10 (x64) [1803]

dotSK, 2018-10-24:
**Description**
Installation fails because of Expand-Archive function parameters change i guess. PowerShell reports that it doesn't know options '-OutputPath' and '-ShowProgress'. Futhermore, because it expects a zip file, a file with .tmp extension is not recognized.

**Expected behaviour**
To successfully install GHDL using PowerShell.

I'll create a pull request with changes.

dotSK, 2018-10-28:
I have the pull request ready fixing this, assuming we choose to go "builtin function" way.

Paebbels, 2018-10-28:
Of cause, you can create a PR. I'll do a review and afterwards merge your proposed changes.

PowerShell incorporated a lot of functionally, that was missing some versions ago. So PSCX was needed to get a good working PS environment.

The idea is:
- Who wants to install GHDL with the PS package/scripts should not need PSCX.
- Who likes to compile GHDL might still need PSCX for e.g. packaging.

closed 2018-10-28 by Paebbels

1138-4EB, 2018-10-29:
@dotSK, just a suggestion: you can use `#457` instead of `[linked issue](https://github.com/ghdl/ghdl/issues/457)` to cross-reference issues in this same repo. Furthermore, you can use `user/repo#000` to reference issues in other github projects.

commit c194a74d (2018-10-28): Merge pull request #681 from dotSK/fix/powershell-windows-install

PowerShell Windows installation fix.
Fixes #681. Fixes #678.