ghdl/ghdl #680: Parsing of type mark do not allow slice or index

go2sh, 2018-10-27:
**Description**
The subtype attribute is allowed on any object and can be used as type mark. So the following should be allowed:
```
constant asd : std_logic_vector(1 downto 0) := "00";
constant asdf : asd(0)'subtype := '1';
```

**Expected behaviour**
Compiles with success.

**Context**
```
library ieee;
use ieee.std_logic_1164.all;

package a is

function b(ads : integer) return std_logic_vector;

constant asdfg : std_logic_vector(1 downto 0) := "00";
end package;

package body a is
function b(ads : integer) return std_logic_vector is
  variable gg : asdfg(1)'subtype;
begin

return "00";
end function;

end package body;
```    
Version:
GHDL 0.36-dev (v0.35-339-g5a97d9f2)

PS: I'am posting this here just for completeness. This language is very strange sometimes. ;-) I don't expect any fix.

Paebbels, 2018-10-28:
The bug reported in #641 uses a similar syntax to declare new objects.

go2sh, 2018-10-28:
This one doesn't even get parsed. Same direction but a different problem.

tgingold, 2018-11-12:
I agree this is allowed by vhdl 08.

tgingold, 2018-11-12:
But I also agree this is awful and makes the VHDL grammar more complex; not anymore LALR I would think.

go2sh, 2018-11-12:
I also came across this, because its hard to parse. I started to create a new tm-grammar and wanted to highlight subtype indications, but with the code above it's quite hard.