ghdl/ghdl #683: GHDL crashes without clear message

obruendl, 2018-10-29:
**Description**
If I run the simulation of our fixed-point library, GHDL crashes. I don't get any error output.

**Context**
There is no `GHDL Bug occurred` block printed. 

I only get:
"This application has requested the Runtome to terminate in an unusual way.
Please contact the application's support team for more information"

- OS: Windows 10
- Environment: I run GHDL from a TCL script

**How to reproduce?**
A TCL interpreter (e.g. wish) is required for reproduction since the simulation is controlled by a TCL script.

1. Clone this repository (use --recursive since it contains submodules)
https://github.com/paulscherrerinstitute/psi_fpga_all
2. Navigate to VHDL/psi_fix/sim
3. delete the *.cf file to clean-up 
4. start the TCL interpreter (e.g. run tclsh)
5. run "source ./runGhdl.tcl"

The script now runs for a while. During the simulation of "psi_fix_dds_18b_tb" I get the error message described further above.

**Files**
https://github.com/paulscherrerinstitute/psi_fpga_all

tgingold, 2018-11-13:
I can reproduce one crash.  Investigating.

tgingold, 2018-11-13:
At en_cl_fix_pkg.vhd:1551, you have:
```
1551				assert not ((saturate = Warn_s or saturate = SatWarn_s) and (a_v >= 2**result_fmt.IntBits or a_v < 0))
```
But `result_fmt.IntBits` is 31, so you are computing 2**31, which is not a valid Integer.
There shouldn't be a crash, but in any case you would have an error.

commit 279fcbc2 (2018-11-14): Fix overflow detection for **, implement ** for i64.

Fix #683
closed 2018-11-14 by tgingold