ghdl/ghdl #684: Are there alternatives for Travis-CI?

Paebbels, 2018-10-29:
I noticed that Travis-CI needs >12h to schedule an open source job in there pipeline.

20h ago I merged a PR.  
7h ago I received the Travis-CI report.

Are they doing this by purpose to open source projects to get more premium customers?

1138-4EB, 2018-10-30:
> Are they doing this by purpose to open source projects to get more premium customers?

I don't think so. Travis is certainly not the most stable platform, but those 12h/7h are definitely exceptional. Precisely, travis did have some problems this week. See Oct 28, Oct 23 and Oct 22: https://www.traviscistatus.com/#month

Moreover, I have found more network issues in the last months. They are changing part of the underlying services: https://blog.travis-ci.com/2018-07-23-the-tale-of-ftp-at-travis-ci

> Are there alternatives for Travis-CI?

I am not aware of any other which is not self-hosted and still offers as much resources as travis. But it might be worth analysing other alternatives. However, note that the feature list is quite complete: docker based GNU/Linux builds, macOS virtual machines, tight-integration with GitHub (checks), good integration with Docker Hub, https://blog.travis-ci.com/2018-10-25-dashboard-my-builds-release... and some other we are not using (yet): https://blog.travis-ci.com/2018-10-11-windows-early-release

Paebbels, 2018-10-30:
Thanks @1138-4EB for the insights. I just wanted to raise these questions.

wfjm, 2018-10-31:
I'm using Travis for my GitHib projects, and so far had rarely to wait for a job start, almost always _instant service_. Not bad for a _free service_ after all. Great system. The main draw back from my view point is that they to this day they offer Trusty (aka Ubuntu 2014.04 LTS) as most up-to-date Linux. It's easy to use newer compilers, but when it comes to update boost for example it quickly gets a bit messy. Given that we live in 2018 and that Ubuntu 2018.04.1 is out, I'd expect that they offer Xenial (aka Ubuntu 2016.04 LTS).

1138-4EB, 2018-10-31:
Hi @wfjm! Thanks for your feedback.

About the issue with the provided OS and version, in this project we work around it with [docker](https://www.docker.com/). We use the OS provided by Travis for macOS only. All the GNU/Linux builds and tests are executed in docker containers (Debian, Ubuntu and Fedora). Precisely, we run tests on Ubuntu 14, Ubuntu 16 and Ubuntu 18 (see e.g. https://travis-ci.com/ghdl/docker/builds/89751237). For completeness, Windows builds and tests are executed on [AppVeyor](https://www.appveyor.com/).

tgingold, 2018-11-04:
Thanks for the useful comments and POV.

closed 2018-11-04 by tgingold