ghdl/ghdl #689: VHDL2008 Unconstrained Array Causes ORTHO_CODE.SYNTAX_ERROR

manjagu, 2018-11-05:
**Description**
The VDHL 2008 unconstrained array constructs are being used in testing with VUnit but fail compilation for unknown reasons. Same file does not fail in ModelSim compilation. Code below. It may be related to #467. Compilation occurs using VUnit python scripts, but VUnit libraries are not required to produce the bug.

**Expected behaviour**
Code should compile without error and indicate all tests pass. Actual result is "GHDL Bug occurred".

**Context**
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: 0.35 (tarball) [Dunoon edition]
Compiled with GNAT Version: GPL 2017 (20170515-63)
Target: i686-pc-mingw32
In directory: E:\git\temporal_video_filter\
Command line:
C:\ghdl\bin\ghdl.exe --elab-run --std=08 --work=lib --workdir=E:\git\temporal_video_filter\vunit_out\ghdl\libraries\lib -PE:\git\temporal_video_filter\vunit_out\ghdl\libraries\vunit_lib -PE:\git\temporal_video_filter\vunit_out\ghdl\libraries\lib tb_example tb -grunner_cfg=active python runner : true,enabled_test_cases : ,output path : E::/git/temporal_video_filter/vunit_out/test_output/lib.tb_example.all_7b5933c73ddb812488c059080644e9fd58c418d9/,tb path : /,use_color : true --assert-level=error
Exception ORTHO_CODE.SYNTAX_ERROR raised
Exception information:
raised ORTHO_CODE.SYNTAX_ERROR : ortho_code-exprs.adb:1055
******************************************************************

- OS: Windows 10 Pro Version 1803
- Origin:
  - [ ] Released binary: https://github.com/ghdl/ghdl/releases/download/v0.35/ghdl-0.35-mcode-windows.zip

**Additional context**
None. Use example code as is.

**How to reproduce?**

1. Compile the example code below using run.py. GHDL bug will be reported. 

2. Comment out the line:
`signal switch_out : my_array(0 to 7)(2 downto 0);`
and uncomment the line:
`signal switch_out : my_array(0 to DDR_DATA_WIDTH / FILTER_WORD_SIZE - 1)(natural(log2(real(DDR_DATA_WIDTH / FILTER_WORD_SIZE))) - 1 downto 0);`

3. Save and re-compile using run.py. No error reported. All tests pass.

```
#>> tb_example.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package test_pkg is
  type my_array is array(natural range <>) of unsigned;
end package test_pkg;

library vunit_lib;
context vunit_lib.vunit_context;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.math_real.all;
use work.test_pkg.all;

entity tb_example is
  generic (runner_cfg : string);
end entity;

architecture tb of tb_example is
  constant DDR_DATA_WIDTH       : integer := 128;
  constant FILTER_WORD_SIZE     : integer := 16;

--  signal switch_out : my_array(0 to DDR_DATA_WIDTH / FILTER_WORD_SIZE - 1)(natural(log2(real(DDR_DATA_WIDTH / FILTER_WORD_SIZE))) - 1 downto 0);
  signal switch_out : my_array(0 to 7)(2 downto 0);
    
begin
  main : process
  begin
    test_runner_setup(runner, runner_cfg);
    report "Hello world!";
    test_runner_cleanup(runner); -- Simulation ends here
    wait;
  end process;
end architecture;`
#>> sim.sh
ghdl -a ent.vhd
ghdl --elab-run ent

#>> run.py
from os.path import join, dirname
from vunit import VUnit

root = dirname(__file__)

ui = VUnit.from_argv()
lib = ui.add_library("lib")
lib.add_source_files(join(root, "*.vhd"))
lib.add_source_files(join(root, "*.vhdl"))
ui.main()

#>> end
```

Paebbels, 2018-11-06:
Please check the attached version of GHDL v0.36-dev: [ghdl-mcode-v0.36-dev.zip](https://github.com/ghdl/ghdl/files/2554036/ghdl-mcode-v0.36-dev.zip)

You are using v0.35. There have been many changes to the unconstrained and partially constrained records since v0.35. Please verify again and report back. Thanks.

manjagu, 2018-11-07:
I'm sorry to report that it did not work. Please see below:
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: @VER@ (tarball) [Dunoon edition]
Compiled with GNAT Version: GPL 2017 (20170515-63)
Target: i686-pc-mingw32
In directory: E:\git\tvf\
Command line:
C:\ghdl\bin\ghdl.exe --elab-run --std=08 --work=lib --workdir=E:\git\tvf\vunit_out\ghdl\libraries\lib -PE:\git\tvf\vunit_out\ghdl\libraries\vunit_lib -PE:\git\tvf\vunit_out\ghdl\libraries\lib data_switch_tb testbench -grunner_cfg=active python runner : true,enabled_test_cases : Checking switch_sel rotation,output path : E::/git/tvf/vunit_out/test_output/lib.data_switch_tb.Checking_switch_sel_rotation_00039a712918017d1e963f81a07af61e60c5afd1/,tb path : /,use_color : true --assert-level=error
Exception ORTHO_CODE.SYNTAX_ERROR raised
Exception information:
raised ORTHO_CODE.SYNTAX_ERROR : ortho_code-exprs.adb:464
******************************************************************

commit 3bd4d200 (2018-11-10): Add testcase from #689

close #689
closed 2018-11-10 by tgingold