ghdl/ghdl #690: cannot handle IIR_KIND_INTERFACE_PROCEDURE_DECLARATION

felixn, 2018-11-07:
**Description**
When compiling an entity with a generic procedure, ghdl throws an internal error:
`sem_subprogram_call_stage1: cannot handle IIR_KIND_INTERFACE_PROCEDURE_DECLARATION (/home/nif3fe/src/dap/src/source_sink/source.vhd:7:15)`

**Expected behaviour**
Expect entity to compile correct, instead ghdl throws an internal error.

**Context**
```
=== Command used: ===
/home/nif3fe/packages/ghdl_mcode/bin/ghdl -a --workdir=/home/nif3fe/src/dap/vunit_out/ghdl/libraries/source_sink --work=source_sink --std=08 -P/home/nif3fe/src/dap/vunit_out/ghdl/libraries/vunit_lib -P/home/nif3fe/src/dap/vunit_out/ghdl/libraries/source_sink /home/nif3fe/src/dap/src/source_sink/source.vhd

=== Command output: ===
/home/nif3fe/src/dap/src/source_sink/source.vhd:27:5:warning: declaration of "data" hides port "data" [-Whide]
/home/nif3fe/src/dap/src/source_sink/source.vhd:28:5:warning: declaration of "valid" hides port "valid" [-Whide]
/home/nif3fe/src/dap/src/source_sink/source.vhd:31:36:warning: declaration of "stimuli" hides file "stimuli" [-Whide]
/home/nif3fe/src/dap/src/source_sink/source.vhd:31:36:warning: declaration of "stimuli" hides file "stimuli" [-Whide]
sem_subprogram_call_stage1: cannot handle IIR_KIND_INTERFACE_PROCEDURE_DECLARATION (/home/nif3fe/src/dap/src/source_sink/source.vhd:7:15)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-343-gc194a74d) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: x86_64-linux-gnu
In directory: /home/nif3fe/src/dap/
Command line:
/home/nif3fe/packages/ghdl_mcode/bin/ghdl -a --workdir=/home/nif3fe/src/dap/vunit_out/ghdl/libraries/source_sink --work=source_sink --std=08 -P/home/nif3fe/src/dap/vunit_out/ghdl/libraries/vunit_lib -P/home/nif3fe/src/dap/vunit_out/ghdl/libraries/source_sink /home/nif3fe/src/dap/src/source_sink/source.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : errorout.adb:246
Call stack traceback locations:
0x55cc853df949 0x55cc853afbec 0x55cc853affff 0x55cc853b0720 0x55cc853883a3 0x55cc8538825f 0x55cc853884cf 0x55cc853bff04 0x55cc85390e1d 0x55cc8538991d 0x55cc853bbcaf 0x55cc853c25c1 0x55cc8537c7b0 0x55cc8550e2ae 0x55cc854fd689 0x55cc8552a4b6 0x55cc85231633 0x7fb3b3d49b95 0x55cc85230458 0xfffffffffffffffe
******************************************************************
```

- OS: Ubuntu 18.04.1 LTS
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [X] Built from sources. Commit SHA: c194a74d

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE), that is compatible with [issue-runner](https://github.com/1138-4EB/issue-runner). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```
#>> source.vhd
library ieee;
use ieee.std_logic_1164.all;

entity source is
  generic(
    type data_type;
    procedure read(l: inout std.textio.line; value: out data_type; good: out boolean);
    stm_file: string
  );
  port(
    clk: in std_logic;
    resetn: in std_logic;

    data: out data_type;
    valid: out std_logic;
    ready: in std_logic;

    valid_i: in std_logic := '1'
  );
end entity source;

architecture behav of source is

  file stimuli: std.textio.text open read_mode is stm_file;

  type packet_t is record
    data: data_type;
    valid: std_logic;
  end record;

  impure function next_packet(file stimuli: std.textio.text) return packet_t is
    variable stimuli_line: std.textio.line;
    variable packet: packet_t;
    variable good: boolean := false;
  begin
    while not std.textio.endfile(stimuli) and not good loop
      std.textio.readline(stimuli, stimuli_line);
      read(stimuli_line, packet.data, good);
    end loop;

    packet.valid := '1' when good else '0';
    
    return packet;
  end function;

  signal packet: packet_t;
  signal init: std_logic;

begin

  process(clk) is
  begin
    if rising_edge(clk) then
      if resetn = '0' then
        packet.valid <= '0';
        init <= '0';
      else
        if init = '0' or (packet.valid = '1' and valid_i = '1' and ready = '1') then
          packet <= next_packet(stimuli);
          init <= '1';
        end if;
      end if;
    end if;
  end process;

  data <= packet.data;
  valid <= packet.valid and valid_i;

end architecture behav;

#>> sim.sh
ghdl -a --std=08 source.vhd

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Checklist**
Before submitting your issue, please review the following checklist:

- [X] Add `GHDL Bug occurred` log block
- [X] Add a MWE
- [X] Try the latest version

Paebbels, 2018-11-07:
While GHDL should not crash and report proper error messages, I'm not sure if this is allowed in VHDL-2008. The datatype `packet_t` and its memory layout is known very late, because it depends on a generic interface type.

tgingold, 2018-11-07:
The error message could be improved, but currently generic subprograms and types are not supported for entities.