ghdl/ghdl #692: Segfault if to_unsigned()'s second parameter is 10 mln

opk12, 2018-11-10:
to_unsigned(integer(0.0), N) segfaults if N is too high.

**Context**
- OS: Debian 10 Buster (currently "testing")
- Origin: Package manager. Repo: Debian 10
- Version 0.35+git20180503+dfsg-2 (current for Debian 10)

**Additional context**
The upper bound is not constant across different runs of ghdl.

In the example file, the "bound check failure" message is not printed because the segfault occurs first.
```
ghdl -r tb --vcd=vcd.vcd --stop-time=800ns
make: *** [Makefile:15: run] Segmentation fault
```

**How to reproduce?**

```
#>> ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity tb is end entity;
architecture arch of tb is
	signal reproducer: unsigned(15 downto 0);
begin
	reproducer <= to_unsigned(integer(0.0), 10**7);
end arch;

#>> sim.sh
ghdl -a ent.vhd
ghdl --elab-run  --stop-time=800ns ent

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

Paebbels, 2018-11-11:
Is the conversion `integer(0.0)` required to reproduce the problem?

tgingold, 2018-11-11:
This is simply a stack overflow.
Should ghdl emits an error if a too large object is allocated ?

tgingold, 2018-11-23:
By adding a dynamic check, ghdl now writes:
```
ghdl_mcode:error: declaration of a too large object (9765 KB)
  from: ieee.numeric_std.to_unsigned at numeric_std-body.vhdl:3074
  from: process work.tb(arch).P0 at ent.vhdl:9
ghdl_mcode:error: simulation failed
```

commit f2c4cfad (2018-11-23): Add --max-stack-alloc option, check stack allocation of complex object.

Fix #692
closed 2018-11-23 by tgingold