ghdl/ghdl #695: Generic array of generic type

DBLouis, 2018-11-16:
**Description**
I want to make a generic multiplexer.

**Expected behaviour**
It does not compile.

**Context**
- OS: Debian buster
- Origin: Debian buster GCC Docker image

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-367-g9d3b26ba) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: x86_64-linux-gnu
In directory: /home/louis/work/
Command line:
/usr/local/libexec/gcc/x86_64-pc-linux-gnu/7.2.0/ghdl1 --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -quiet -o multiplexer.s multiplexer.vhdl
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : sem_inst.adb:81 index check failed
Call stack traceback locations:
0x66bb59 0x66df77 0x6452fc 0x66b76b 0x63b863 0x6bc4bc 0x6bb357 0x5f6228 0xabdbed 0x5f05f7 0x5fafc9 0x725969 0x5eeb98 0x7f76d5978b15 0x5f26c8 0xfffffffffffffffe
******************************************************************
```

```
#>> mux.vhd
library ieee;
use ieee.std_logic_1164.all;

package mux is
  generic (
    MUX_DATA_SIZE : natural
  );
  subtype mux_data is std_logic_vector(MUX_DATA_SIZE-1 downto 0);
  type mux_data_array is array (natural range <>) of mux_data;
end package;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity multiplexer is
  generic (
    MUX_DATA_SIZE : natural;
    MUX_CTRL_SIZE : natural;
    package mux_g is new work.mux generic map (MUX_DATA_SIZE => MUX_DATA_SIZE)
  );
  port (
    MUX_CTRL : in std_logic_vector(MUX_CTRL_SIZE-1 downto 0);
    MUX_IN : in mux_g.mux_data_array(0 to 2**MUX_CTRL_SIZE-1);
    MUX_OUT : out mux_g.mux_data
  );
end entity;

architecture multiplexer_arch of multiplexer is
begin
  MUX_OUT <= MUX_IN(to_integer(unsigned(MUX_CTRL)));
end architecture;

#>> sim.sh
ghdl -a mux.vhd
ghdl --elab-run mux

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-gcc-7.2.0 sh -c ./sim.sh

#>> end
```

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

tgingold, 2018-11-17:
Generic map aspect in interface package declaration are not yet supported.

tgingold, 2018-11-17:
In addition, interface package declarations are also not yet supported in entities...

DBLouis, 2018-11-17:
Is there any way I can achieve what I want to do?

tgingold, 2018-11-17:
You can use an unbounded array of std_logic_vector (but the support of it is still WIP in ghdl).

Paebbels, 2018-11-17:
Please see [my answer on CodeReview](https://codereview.stackexchange.com/questions/73708/vhdl-mux-in-need-of-generics) about 4 different solutions on generic multiplexers for the `std_logic` datatype.

If you also want to have the type generic, you may need to wait for VHDL-2008 and partially for [VHDL-2018](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/VHDL2017) support in tools.

eine, 2021-01-17:
I can analyze but not elaborate the provided MWE with a recent version of GHDL:

```sh
# ghdl --version
GHDL 1.0-dev (0.37.0.r1270.gb65ee8c3) [Dunoon edition]
 Compiled with GNAT Version: 10.2.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2020 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

# ghdl -e --std=08 mux
error: "mux" is neither an entity nor a configuration
C:\msys64\mingw64\bin\ghdl.exe: compilation error

# ghdl -e --std=08 multiplexer
error: entity "multiplexer" cannot be at the top of a design
mux.vhd:20:13: (package interface "mux_g" is a package generic)
C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

tgingold, 2021-01-17:
Yes, the reproducer is incomplete.

closed 2023-12-17 by tgingold