ghdl/ghdl #696: Help with design low-level HDL language

XVilka, 2018-11-16:
FPGA world suffers a lot from fragmentation - some tools produce Verilog, some VHDL, some - only subsets of them, creating low-level LLVM-like alternative will help everyone, so HDL implementations will opt only for generating this low-level HDL and routing/synthesizers accept it. LLVM or WebAssembly - you can see how many languages and targets are supported now by both. With more open source tools for FPGA this is more feasible now than ever.

See SymbiFlow/ideas#19

Paebbels, 2018-11-16:
I don't get your message here.

I think I have not understood how hardware designing works ...  
At first, Verilog and VHDL are input languages and normally not the output of tools.  
At second, the wished exchange format already exists: it's a netlist.  
Netlists can be described in different file formats.

At next, the translation of HDL to a netlist isn't called compilation. It's synthesis, because a frontend does more then just simple translation. It infers a circuit description from a hardware description encode in a simulation language. Thus the synthesizer is part of the frontend and not of the backend.

XVilka, 2018-11-16:
I called it compilation to show the analogy with LLVM. Of course FPGA pipeline is very different and more complex, but the general idea is still the same. Problem with netlists - there are different formats for different vendors or even tools. My idea is to create something in between. See the answers to the original issue: https://github.com/SymbiFlow/ideas/issues/19#issuecomment-439409548 and https://github.com/SymbiFlow/ideas/issues/19#issuecomment-439466125

closed 2021-09-07 by tgingold