ghdl/ghdl #697: Generic package instantiation failed

DBLouis, 2018-11-18:
**Description**
Generic package with subtypes.

**Expected behaviour**
It fails to compile.

**Context**

- OS: Debian buster
- Origin: Debian buster GCC Docker image

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-364-g2fc9b6a8) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: x86_64-linux-gnu
In directory: /home/louis/work/
Command line:
/usr/local/libexec/gcc/x86_64-pc-linux-gnu/7.2.0/ghdl1 --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -quiet -o register_file.s register_file.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : sem_decls.adb:429
Call stack traceback locations:
0x645a99 0x66b76b 0x63b863 0x6bc4bc 0x6bb357 0x5f6228 0xabd9bd 0x5f05f7 0x5fafc9 0x725739 0x5eeb98 0x7f001ad70b15 0x5f26c8 0xfffffffffffffffe
******************************************************************
```

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE), that is compatible with [issue-runner](https://github.com/1138-4EB/issue-runner). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```
#>> ent.vhd
library ieee;
use ieee.std_logic_1164.all;

package p_register is
  generic (
    ADDR_BITS : positive;
    DATA_BITS : positive
  );
  subtype vec_addr is std_logic_vector(ADDR_BITS-1 downto 0);
  subtype vec_data is std_logic_vector(DATA_BITS-1 downto 0);
end package;

--
library ieee;
use ieee.math_real.all;
use ieee.std_logic_1164.all;

entity e_register_file is
  generic (
    RF_COUNT : positive;
    RF_DATA_BITS : positive;
    constant RF_ADDR_BITS : positive := positive(ceil(log2(real(RF_COUNT))));
    package pr is new work.p_register generic map (RF_ADDR_BITS, RF_DATA_BITS)
  );
  port (
    CLK : in std_logic;
    RF_WRITE : in std_logic;
    RF_READ_ADDR_1 : in pr.vec_addr;
    RF_READ_ADDR_2 : in pr.vec_addr;
    RF_WRITE_ADDR : in pr.vec_addr;
    RF_IN : in pr.vec_data;
    RF_OUT_1 : out pr.vec_data;
    RF_OUT_2 : out pr.vec_data
  );
end entity;

architecture a_register_file of e_register_file is
  type t_reg_array is array (0 to RF_COUNT-1) of pr.vec_data;
  signal REGISTERS_IN : t_reg_array;
  signal REGISTERS_OUT : t_reg_array;
  signal REGISTERS_WRITE : std_logic_vector(t_reg_array'range);
  function addr_to_index (a : pr.vec_addr) return integer is
  begin
    return to_integer(unsigned(a));
  end addr_to_index;
begin
  gen: for i in t_reg_array'range generate
    reg: entity work.e_register
      generic map (RF_DATA_BITS)
      port map (
        CLK => CLK,
        REGISTERS_WRITE(i) => REG_WRITE,
        REGISTERS_IN(i) => REG_IN,
        REGISTERS_OUT(i) => REG_OUT
      );
  end generate;
  REGISTERS_IN(addr_to_index(RF_WRITE_ADDR)) <= RF_IN when rising_edge(CLK) and RF_WRITE = '1';
  REGISTERS_WRITE <= std_logic_vector(to_unsigned(2**addr_to_index(RF_WRITE_ADDR), RF_COUNT)) when rising_edge(CLK) and RF_WRITE = '1';
  RF_OUT_1 <= REGISTERS_OUT(addr_to_index(RF_READ_ADDR_1)) when rising_edge(CLK) else (others => 'Z');
  RF_OUT_2 <= REGISTERS_OUT(addr_to_index(RF_READ_ADDR_2)) when rising_edge(CLK) else (others => 'Z');
end architecture;

#>> sim.sh
ghdl -a ent.vhd
ghdl --elab-run ent

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-gcc-7.2.0 sh -c ./sim.sh

#>> end
```

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

doppioandante, 2018-11-18:
I was just about to submit a similar testcase:
```vhdl
library ieee;
use ieee.std_logic_1164.all;

package a is
    generic(
        size: positive
    );
    subtype t is std_logic_vector(size-1 downto 0);
end package a;

entity test is
end test;

architecture dataflow of test is
    package p is new work.a generic map(
        size => 3
    );
begin
    entities:
    for i in 0 to 7 generate
         signal a: work.a.t;
    begin
    end generate entities;
end dataflow;
```

tgingold, 2018-11-18:
This last one is interesting as it is not valid.

doppioandante, 2018-11-18:
> This last one is interesting as it is not valid.

Interesting, why is that?

commit 237cc1e9 (2018-11-19): Check for uninstantiated package in expanded names.

Fix #697
closed 2018-11-19 by tgingold