ghdl/ghdl #698: Stack Overflow

dtp555-1212, 2018-11-18:
GHDL release: 0.35 (tarball) [Dunoon edition]
Compiled with GNAT Version: 4.9.3
Target: x86_64-linux-gnu
In directory: /home/user/LocalShare/vhdl/junk/
Command line:
ghdl -c -e junk
Exception STORAGE_ERROR raised
Exception information:
Exception name: STORAGE_ERROR
Message: stack overflow or erroneous memory access
Call stack traceback locations:
0x7f873c0634ae 0x7f873c408312 0x63765b 0x638b8c 0x638f91 0x6454ff 0x64566b 0x625bd3 0x62a319 0x674854 0x618043 0x672da3 0x62273a 0x677659 0x68013d 0x67dfd3 0x4ed9d3 0x40964f 0x7f873c04e82e 0x408427 0xfffffffffffffffe

Same file seems to be digestable in Vivado, but crashes in ghdl in about 10 seconds.

Saw an earlier report of same issue in an older version, but issue was marked closed as they could not recreate

See attached for file to use to recreate
[vhdljunk.zip](https://github.com/ghdl/ghdl/files/2593024/vhdljunk.zip)

tgingold, 2018-11-18:
It's a huge case statement with lines like this:
```
when "01010100000000000000000000000" => o <= X"4";
when "01010101000000000000000000000" => o <= X"5";
```

tgingold, 2018-11-18:
A large array is created on the stack, which causes a stack overflow.

tgingold, 2018-11-20:
Now works for me (with mcode).  But still, it's a lot of code...  that could be factorized.

closed 2018-11-20 by tgingold