ghdl/ghdl #699: Naming of library files

Paebbels, 2018-11-18:
<s>All</s> Many *body* files are named `***-body`, whereas `std_logic_1164_body.vhdl` is using a underline character.
Is this a typo or an inconsistency in the standard?

Files with underline:
* `std/env_body.vhdl`
* `std/textio_body.vhdl`
* `ieee/std_logic_1164_body.vhdl`
* `mentor/std_logic_arith_body.vhdl`
* `vital95/vital_primitives_body.vhdl`
* `vital95/vital_timing_body.vhdl`

tgingold, 2018-11-18:
I think the standard is silent about those names.
But yes it would be nice to be consistent and use `-body`.

Paebbels, 2018-11-18:
As a followup question:

Why are sources of library `IEEE` stored in the folders `ieee` and `ieee2008`. I always though the special comments like `--V87` are used to generate the required package lines from one source file.

maehne, 2019-04-24:
I would also appreciate if the naming of the library files could be made more consistent as this eases setting up a `vhdl_ls.toml` project file for online syntax checking using the `vhdl_ls` language server developed as part of the [kraigher/rust_hdl](https://github.com/kraigher/rust_hdl/) project.

tgingold, 2019-04-24:
Yes, that makes sense.

commit f4689c26 (2019-04-25): libraries: rename _body files.

Fix #699
closed 2019-04-25 by tgingold