ghdl/ghdl #700: Sin() mismatch

obruendl, 2018-11-20:
**Description**
The results of my simulations in Modelsim and GHDL do not match. I tracked down the issue a bit and found out that the result of the coefficient calculation in VHDL (based on the sin()-function) returns different values in Modelsim and GHDL.
In our setup we compare the output of the HDL simulations to bittrue python models. Python produces the same results as Modelsim, so I consider the Modelsim sin() implementation as "correct".

**Expected behaviour**
I would expect the sin() function to return the same value for Modelsim and GHDL. Maybe in some cases they differ by rounding, so one "LSB" (it's float so this term is a bit strange) of difference would be acceptable but not more. 
Since coefficients are rounded to 25 bits and real usually is implemented as double, I do not expect to see any rounding effects in the coefficient values.

**Context**
No crash, so no context

OS: Windows 10
Environment: I run GHDL from a TCL script
How to reproduce?
A TCL interpreter (e.g. wish) is required for reproduction since the simulation is controlled by a TCL script.

**How to reproduce?**

1. Clone this repository (use --recursive since it contains submodules)
https://github.com/paulscherrerinstitute/psi_fpga_all
2. For the repository "<root>/VHDL/en_cl_fix" checkout the latest tag (1.1.1). This step is required to fix #683 which prevents the simulation to reach the point described in this bug report).
3. Navigate to VHDL/psi_fix/sim
4. delete the *.cf file to clean-up
5. start the TCL interpreter (e.g. run tclsh)
6. run "source ./runGhdl.tcl"
7. The script now runs for a while. During the simulation of "psi_fix_mod_cplx2real_tb" you will see errors reported by the effect described above.
8. To run the modelsim version of the simulation, run Modelsim, navigate to VHDL/psi_fix/sim and run "source ./run.tcl" in the modelsim TCL console. You should not see any errors.

The code that behaves differently is located in VHDL/psi_fix/hdl/psi_fix_mod_cplx2real.vhd, lines 61 ... 80.

Files
https://github.com/paulscherrerinstitute/psi_fpga_all

tgingold, 2018-11-24:
I suppose it would work with openieee, that uses libm to compute sin().

obruendl, 2018-12-14:
How can I use openieee? As far as I could find out, it looks like openieee was developed as part of GHDL, right? Is there just a flag to pass to say which ieee implementations I want to use? Or do I have to download and compile openieee separately? 
What IEEE implementation does GHDL use by default (obviously it is not openieee)?

tgingold, 2018-12-15:
Right now you need to rebuild using --enable-openieee.
I have to investigate if that could be the default.
By default GHDL uses the IEEE implementation.