ghdl/ghdl #701: Issue during install

Defossez, 2018-11-22:
Hi,

Using:
    Ubuntu 18.04
   Gnat 2018 latest version installed.
    Install Ghdl in /opt/Ghdl
During make install one of the lines/comments in the terminal is:
    install: cannot stat 'lib/ghdl/src/openieee/*': No such file or directory
The install continues and ends.
Seems everything is installed.

Questions:
    Is the comment line something to worry about?
    Do I need to install something else?
    Will Ghdl work as supposed?

Many thanks,

Marc

Paebbels, 2018-11-22:
GHDL will work as expected. No need to install anything else.

@tgingold can you have a look at that line?

Defossez, 2018-11-23:
Thanks Paebbels!
I tried it and it works as expected.

I think theline has to do with the fact Ghdl is not installed in the default location.

commit 534fd656 (2018-11-24): make install: copy openieee library only if enabled.

Fix #701
closed 2018-11-24 by tgingold