ghdl/ghdl #702: GHDL mingw64-llvm35 fails on AppVeyor

Paebbels, 2018-11-22:
**Description**
The build for platform MinGW64 + LLVM 3.5 fails on AppVeyor.

**Expected behavior**
Do not fail :).

**Context**
[image]

- OS: Windows Server 2016
- Origin: Commit SHA: many latest commits

**Additional context**
MinGW64 has updated LLVM to another version, so LLVM 3.5 is not available anymore.

Paebbels, 2018-11-26:
According to my local MinGW64 installation, the default LLVM version is 7.0.

    Patrick Lehmann@PLC2-Paebbels MINGW64 ~
    $ llvm-config.exe --version
    7.0.0

@1138-4EB  
Have I missed something? When and why did LLVM jump to 7.0?

@tgingold  
Does GHDL already support LLVM 7.0?

Paebbels, 2018-11-26:
LLVM 7.0 support is proposed in #703.

1138-4EB, 2018-11-26:
@Paebbels, IDK. They jumped from 5 to 7 in less than a year. However, Fedora 29 (which was released in October) is using LLVM 6 yet.

BTW, it is almost a year since the last GHDL release was published. @tgingold, @Paebbels, should we consider releasing v0.36 in December?

Paebbels, 2018-11-26:
@tgingold is working on the unconstrained record stuff to get it finished, so we can announce, *full* support for UVVM and OSVVM AXI models.

tgingold, 2018-11-28:
As far as I can see, there are still failures on Appveyor.  Not a build failure, but some tests fail.

1138-4EB, 2018-11-29:
@Paebbels do you have those test failures on your machine? I'm trying to tell if we have a problem with appveyor only, or with LLVM on Windows.

Paebbels, 2018-11-30:
Yes, I need to run the test environment on my machine. As far as I remember, this was not so easy ...

PR #704 (2018-11-26): Removed explicit llvm35 depdendency
**Description**  
MinGW changed from `*llvm35` and `*clang35` packages to `*llvm` and `clang` packages.

This PR removes the version number for LLVM. Latest version used by MinGW is `7.0.0-1`.

This should fix #702.