ghdl/ghdl #707: Crash when assigning array aggregate

LukasVik, 2018-11-26:
**Description**
The following code crashes ghdl elab run:
```
  subtype my_low_rng is integer range 0 to 1;
  subtype my_high_rng is integer range 2 to 3;

  constant my_bad_booleans : boolean_vector(0 to 3) :=
    (my_low_rng => true, my_high_rng => false);
```
This crash was introduced in ee9f2600 and is still present on master HEAD.

**Expected behaviour**
I have no idea about the VHDL standard so maybe this isn't supposed to work. However I consider it a neat way of writing vectors so I would prefer if it did work.

**Context**
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-436-gd570ee98) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: x86_64-linux-gnu
In directory: /src/
Command line:
ghdl -e --std=08 ent
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap3.adb:3056 access check failed
Call stack traceback locations:
0x55b9bc283ed3 0x55b9bc24cc75 0x55b9bc24d5e2 0x55b9bc24ceaa 0x55b9bc24e8c2 0x55b9bc24e989 0x55b9bc235d9f 0x55b9bc236107 0x55b9bc236206 0x55b9bc23d5bf 0x55b9bc22d37b 0x55b9bc2919dc 0x55b9bc1fec35 0x55b9bc1fbfe5 0x55b9bc209b13 0x55b9bc294937 0x55b9bc1e136d 0x55b9bc168936 0x55b9bc2976a2 0x55b9bbf9d2a1 0x7fcbefbecb15 0x55b9bbf9c098 0xfffffffffffffffe
******************************************************************
```

- OS: Ubuntu 18.04 
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: ee9f2600

**How to reproduce?**
MWE below:

```
#>> ent.vhd
entity ent is
end entity;

architecture a of ent is
  subtype my_low_rng is integer range 0 to 1;
  subtype my_high_rng is integer range 2 to 3;

  constant my_good_booleans : boolean_vector(0 to 3) :=
    (0 to 1 => true, 2 to 3 => false);
  constant my_bad_booleans : boolean_vector(0 to 3) :=
    (my_low_rng => true, my_high_rng => false);
begin

  process begin
    report "Hello world" severity note;
    wait;
  end process;

end;

#>> sim.sh
ghdl -a --std=08 ent.vhd
ghdl -e --std=08 ent
ghdl -r --std=08 ent

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

tgingold, 2018-11-26:
Good catch.  I am pretty I have tests for that, but maybe not exercised when choosing vhdl 2008.
I have a fix, I will push it soon.

Thanks!

LukasVik, 2018-11-27:
Great. Thanks!

I noticed later that the some problem occurs when assigning a signal in the same way:
```
architecture a of ent is
  subtype my_low_rng is integer range 0 to 1;
  subtype my_high_rng is integer range 2 to 3;

  signal my_booleans : boolean_vector(0 to 3);
begin
  my_booleans <= (my_low_rng  => true, my_high_rng  => false);
```
So the problem is not limited to only constants.

tgingold, 2018-11-27:
This was a problem in aggregates.

commit b6becbb0 (2018-11-27): sem_expr: fix regression when replace a choice by expression by a choice by range.

Fix #707
closed 2018-11-27 by tgingold