ghdl/ghdl #709: Possibility to catch division by zero?

FranzForstmayr, 2018-11-27:
**Is your feature request related to a problem? Please describe.**
Sometimes i accidentally write a division by zero in a loop, is there a way to get the line number of the affected code?

**Describe the solution you'd like**
Show line number: 
Division by zero at test.vhd:123

**Reproducer**
```
library IEEE;
use     IEEE.std_logic_1164.all;

entity ent is
end entity;

architecture a of ent is
    signal result : integer;
    signal clk: std_logic := '0';
begin
    clk <= not clk after 10 ns;
    process(clk)
        variable i : natural range 0 to 127 := 0;
    begin
        if rising_edge(clk) then
            result <= 10 / i;
        end if;
    end process;

end architecture;
```

**GHDL Message now**
Execution terminated by unhandled exception
raised CONSTRAINT_ERROR : SIGFPE
Call stack traceback locations:
0x7fca45a6d88e 0x7fca46e10106

tgingold, 2018-11-28:
This is supported through backtraces which depends on your configuration.  So which CPU are you using and which backend ?

FranzForstmayr, 2018-11-28:
I use llvm backend most of the time, on a i5-6200U CPU, but on some other machines i have a mcode backend. I haven't tried backtraces yet.

tgingold, 2018-11-30:
So I suppose you are using x86-64.  Correct ?

FranzForstmayr, 2018-11-30:
Yeah