ghdl/ghdl #710: Visibility rules continued

kraigher, 2018-11-28:
```vhdl
package pkg is
  procedure foo;
end package;

package body pkg is
  procedure foo is
  begin
    report "pkg foo";
  end procedure;
end package body;

entity ent is
end entity;

architecture a of ent is
  procedure foo is
  begin
    report "arch foo";
  end procedure;
begin
  main : process
    use work.pkg.foo;
  begin
    foo; -- Causes ambiguity error but not if use clause is moved to architecture declarative region.
  end process;
end architecture;
```

I am not sure what the standard really says regarding this. Why should it work with the use clause in the architecture declarative part but not in a nested declarative part?

kraigher, 2018-11-28:
This seems to not cause an ambiguity error but in both this case and the above the use clause is not within the same immediate region as the declarations of foo.
```vhdl
package pkg is
  procedure foo;
end package;

package body pkg is
  procedure foo is
  begin
    report "pkg foo";
  end procedure;
end package body;

entity ent is
end entity;

architecture a of tb_example is
  use work.pkg.foo;
begin
  blk: block
    procedure foo is
    begin
      report "arch foo";
    end procedure;
  begin
    main : process
    begin
      foo;
    end process;
  end block;
end architecture;
```

tgingold, 2018-11-28:
For the second case: the declaration of foo hides the one that was made visible by the use clause.
For the first case: good catch.  At some point I was confused between immediate declarative region and immediate scope.

The main rule is that declarations imported by use clauses always have a lesser priority than a normal declaration.

commit d142b93c (2018-11-29): sem_scopes: potentially visible name cannot conflict with a normal one.

Fix #710
closed 2018-11-29 by tgingold