ghdl/ghdl #711: Problems with Lattice vendor lib regarding ecp5um

patrickbssp, 2018-11-28:
Dear all

I'm not sure, whether it is appropriate to file my problem as a GHDL bug report, as it may originate from vendor side...

First of all, I appreciate GHDL as a very useful tool for FPGA development. However, I've some troubles to use the Lattice vendor library for the ECP5 familiy and I'm running out of ideas to work around those.

The problem is, that all IPs generated by Clarity Designer (ex IP Express) are referring to components in the library ecp5um, e.g.:

```
library ecp5u;
use ecp5u.components.all;
```

whereas the most appropriate device to be selected in compile-lattice.sh is the ecp5u (note the missing suffix "m"). So, running the script with:

`/opt/ghdl/lib/ghdl/vendors/compile-lattice.sh --out lattice -d ecp5u --vhdl2008`

will create the object files under ./lattice/ecp5u/v08/ as expected, but none of those will be found when elaborating the VHDL files with option -Plattice. Temporarily replacing ecp5u with ecp5um in the use library clause will get me a bit further, just to fail on looking up basic components like DP16KD which are defined in cae_library/synthesis/vhdl/ecp5um.vhd. The component DP16KD is not contained in ecp5u-obj08.cf. The package components is defined in ECP5UCOMP.vhd and the corresponding entry in ecp5u-obj08.cf:

```
file / "/opt/diamond/3.9_x64/cae_library/simulation/vhdl/ecp5u/src/ECP5UCOMP.vhd" "88df0c89f719cfae3ff81d6767d3492082d42055" "20181128153858.876":
  package components at 26( 909) + 0 on 3263 body;
  package body components at 2564( 95278) + 0 on 3264;
```

Any idea on how to circumvent this problem?

Best regards,

Patrick

Paebbels, 2018-11-28:
As I created the scripts for Lattice and ECP5 devices, there was no `ecp5um` version on my disk. I think, you just need to extend the `compile-lattice.sh` file to also support the `um` versions.

Please have a look at:
* [ ] line 48: `DeviceList="... ecp5u ..."`
* [ ] line 243: `FileLists[ecp5u]="..."`

The files need to be listed in the right compile order.

patrickbssp, 2018-11-28:
Thanks for the quick reply!

Unfortunately, there is no folder called ecp5um under .../cae_library/simulation/vhdl/, only ecp5u.

I also tried to symlink the ecp5um to ecp5u, added the device to DeviceList and created a Filelists[ecp5um] entry, but to no avail. I still get the error, that the component can't be bound.

hackfin, 2018-11-29:
Hi Patrick,

the general problem might be that the ECP5 simulation files are incomplete from the VHDL side, so in fact quite a few simulation models are missing in VHDL. It seems Lattice assumes everyone using a mixed language simulator.
You may get a bit further by doing the conversion yourself using icarus verilog, however this requires editing of the .v source in most cases to make it translate ok. In some cases you might even want to write your own emulation (derived from similar ECP3 files).

My  'make' conversion rule:

```
VFLAGS = -Dmixed_hdl -yglobal
VDEPS = gsr_pur_assign.v

%.vhd: $(SRCDIR)/%.v
	iverilog -tvhdl -o $@ -pdepth=1 -s $(patsubst %.vhd,%,$@) \
		$(VFLAGS) $< $(VDEPS)

```

I've already reported this to Lattice Semi a while ago, but it seemed they weren't interested in completing the ECP5 VHDL simulation files. Hopefully that's gonna change, as the ECP5 would be a nice platform to work with.

patrickbssp, 2018-11-29:
Hi hackfin

Thanks for explaining the background. I'll have a look at icarus verilog and hope that I'm able to solve the issue with the help you provided. The ECP5 is quite interesting indeed and it's a pity, that Lattice Semi is so reluctant...

Thanks again!

Patrick

hackfin, 2019-01-07:
Follow up on that one: I got official word from LSemi, bottomline quote:
> We do not support GHDL, only Aldec and Modelsim.
> ...
> There is no plan to support any more VHDL models in Diamond or Radiant

Anyhow, the problem is not related to GHDL whatsoever, newer FPGA architectures (ECP5, LIFMD, ...) are simply lacking VHDL models (architectures of entities, not just component stubs) in source.

tgingold, 2019-01-08:
So we need a mixed vhdl + systemverilog simulator, don't we ?

Paebbels, 2019-01-08:
The `verilog`directory in my Diamond 3.10 installation looks as empty regarding `ecp5um` devices as for VHDL. I can't see SystemVerilog.

arcturus140, 2019-02-13:
@Paebbels any chance you are using a free license for Diamond? `ecp5um` devices require a paid license subscription. 

Has been a while for me looking into Diamond software. Is the folder structure different in Linux / Windows or has it changed recently? I think I am missing `ovi_*` folders for `cae_library` in my linux installation.

Paebbels, 2019-02-13:
@arcturus140 yes, I have a free license, but the installed simulation and synthesis models are not dependent on the license. The license is added after installation. There are simply no files for ECP5UM.

arcturus140, 2019-02-13:
Yes, that makes sense.

arcturus140, 2019-02-14:
I had a look into the build script. 
Seems like pmi are not compiled, as well as verilog primitives where it's VHDL version is missing.
```
# Lattice device libraries
# ==============================================================================
# Excluded: pmi
```

Using the `iverilog -tvhdl` trick mentioned here would you recommend to compile these sources into a library myself or add sources to my GHDL simulation project?

Has there been done some work on this so I can know which files are missing?

hackfin, 2019-02-15:
Hi Arcturus,

I can tell you it's a lot of work to put together a decent library with converted Verilog.
For one thing, many simulation models don't translate correctly. I've gotten a bit further with implementing experimental support for _real_ data types (which occur every now and then), but the biggest concern is that iverilog does not support the translation of Verilog parameters to VHDL generics and would require an architectural change in order to do that.
And, once you've done all the manual conversion, I wouldn't know about the legal grounds about publishing all your work.
You can have this problem with other vendors as well, depending on the FPGA architecture you're targetting. If a full VHDL verification is required, you need to carefully select your chips.

SolidTux, 2020-06-26:
There is a `cae_library/vhdl_packages/vdbs/ecp5um/components.vdb` file (I don't know if this is new). Is this of any use? Or is this just for the synthesis?

hackfin, 2020-07-14:
If I'm informed right, this VDB is a netlist database file you can't use for simulation.
With most of the existing *.v files though, Co-Simulation is possible using OpenSource tools. I've tested a number of ECP5 primitive instancing scenarios run through yosys (post-technology mapping) with the iverilog simulator without major hickups.