ghdl/ghdl #713: ghdl stalls for specific code

slaweksiluk, 2018-11-30:
**Description**
When I run test bench it stops at some point and ctrl+c is needed to exit.

**Expected behaviour**
I expected ghdl to finish test bench after 100 ns.

**Context**
No error message, just stalls.

- OS: Fedora 27
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: 5e5355d8

**Additional context**
I don't know if it is ok in general to ghdl stop test bench during execution like here. However this reproducer behaves strange, as it is enough to change below lines to make it run all the test.
```vhdl
for i in byte_range(desc)'range loop
```
to
```vhdl
for i in 0 to 1 loop
```
or 
```vhdl
variable v : bit_vector(desc.address to desc.address + desc.length);
```
to
```vhdl
variable v : bit_vector(0 to 1);
```
to 

**How to reproduce?**
Requires vunit.  It might be quite hard to extract this reproducer from vunit library, but I can try if it will help you fix it. After first run standalone executable is available in `vunit_out/test_output/lib.dma_controller_tb.bug_9a5d3d925313a0af4bc542063325e04583af3209/ghdl` and can be executed without arguments. Example stack trace from gdb:
```
(gdb) where
#0  0x00000000004b7c9c in vunit_lib__integer_vector_ptr_pkg__integer_vector_ptr_storage_t__set (PARAMS=0xa30890, OBJ=0xa17410)
    at /home/slawek/git/github/vunit/vunit/vhdl/data_types/src/integer_vector_ptr_pkg-body-200x.vhd:63
#1  0x00000000004b8f88 in vunit_lib__integer_vector_ptr_pkg__set (PARAMS=0xa30860) at /home/slawek/git/github/vunit/vunit/vhdl/data_types/src/integer_vector_ptr_pkg-body-200x.vhd:128
#2  0x000000000067a6fd in vunit_lib__memory_pkg__set_permissions (PARAMS=0xa172b0) at /home/slawek/git/github/vunit/vunit/vhdl/verification_components/src/memory_pkg-body.vhd:300
#3  0x000000000068793f in lib__dma_controller_tb__ARCH__bench__mm2s_buffer_destroy__PROC (INSTANCE=0xa16ee0) at /home/slawek/git/flash/SIROCCO-HDL/hdl/test/dma_controller_tb.vhd:83
#4  0x0000000000702b4d in grt.processes.run_processes (postponed=false) at /home/slawek/git/github/ghdl/src/grt/grt-processes.adb:797
#5  0x0000000000703072 in grt.processes.simulation_cycle () at /home/slawek/git/github/ghdl/src/grt/grt-processes.adb:993
#6  0x000000000070341f in grt.processes.simulation_main_loop () at /home/slawek/git/github/ghdl/src/grt/grt-processes.adb:1141
#7  0x0000000000703485 in grt.processes.simulation () at /home/slawek/git/github/ghdl/src/grt/grt-processes.adb:1181
#8  0x000000000070dd16 in __ghdl_run_through_longjump (func=0x703470 <grt.processes.simulation>) at /home/slawek/git/github/ghdl/build/../src/grt/config/jumps.c:232
#9  0x000000000070d9dc in grt.main.run_simul () at /home/slawek/git/github/ghdl/src/grt/grt-main.adb:177
#10 0x000000000070daa1 in grt.main.run () at /home/slawek/git/github/ghdl/src/grt/grt-main.adb:210
#11 0x000000000071dcca in ghdl_main (argc=1, argv=0x7fffffffd9c8) at /home/slawek/git/github/ghdl/src/grt/ghdl_main.adb:59
#12 0x000000000070e006 in main (argc=1, argv=0x7fffffffd9c8) at /home/slawek/git/github/ghdl/build/../src/grt/main.adb:31
```

```vhdl
#>> ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library vunit_lib;
context vunit_lib.vunit_context;
context vunit_lib.com_context;
use vunit_lib.memory_pkg.all;
use vunit_lib.wishbone_pkg.all;
use vunit_lib.bus_master_pkg.all;

use work.sim_codecs_pkg.all;
use work.sim_types_pkg.all;

entity dma_controller_tb is
generic (
	runner_cfg : string := ""
);
end entity;

architecture bench of dma_controller_tb is

	constant num_desc : positive := 4;
	subtype num_desc_range is natural range 0 to num_desc-1;

	constant MEM_BYTES : positive := 512;
	constant MEM_DWORDS : positive := MEM_BYTES/4;

	signal clk : std_logic := '0';
	signal stb : std_logic;
	signal ack : std_logic;
	signal stall : std_logic;
	signal adr : std_logic_vector(31 downto 0);
	signal sel : std_logic_vector(3 downto 0);
	signal dat : std_logic_vector(31 downto 0);

	constant mm2s_memory : memory_t := new_memory;
	constant mm2s_buf : buffer_t := allocate(mm2s_memory, MEM_BYTES);
	constant master_logger : logger_t := get_logger("wb_master");
	constant bus_handle : bus_master_t := new_bus(data_length => 32,
		address_length => 32, logger => master_logger);
	constant CLK_PERIOD : time := 10 ns;

	function byte_range (desc : descriptor_t) return bit_vector is
		-- Works if 0 to 1
		-- variable v : bit_vector(0 to 1);
		variable v : bit_vector(desc.address to desc.address + desc.length);
	begin
		return v;
	end function;

begin

	stimulus : process
	begin
		test_runner_setup(runner, runner_cfg);
		set_format(display_handler, verbose, true);
		show(com_logger, display_handler, verbose);
		wait for 30 ns;

		if run("bug") then
			for i in 0 to MEM_DWORDS -1 loop
				write_word(mm2s_memory, i*4, std_ulogic_vector(to_unsigned(i, 32)));
			end loop;
			write_bus(net, bus_handle, 8, std_logic_vector(to_unsigned(8, 32)));
			wait_until_idle(net, bus_handle);
		end if;
	    test_runner_cleanup(runner);
	end process;

	stop_proc: process begin
		wait for 100 ns;
		std.env.finish;
	end process;

	mm2s_buffer_destroy: process
		variable desc : descriptor_t := (8, 8);
	begin
		wait for 50 ns;
		for i in byte_range(desc)'range loop
		-- Works for below
		--for i in 0 to 1 loop
			set_permissions(mm2s_memory, i, no_access);
		end loop;
	end process;

	wb_master_vc : entity vunit_lib.wishbone_master
	generic map (
		bus_handle => bus_handle,
		strobe_high_probability => 1.0)
	port map (
		clk   => clk,
		adr   => adr,
		dat_i => dat,
		dat_o => dat,
		sel   => sel,
		stb   => stb,
		stall => stall,
		ack   => ack
	);

	wb_slave: process begin
		stall <= '0';
		wait until rising_edge(clk) and stb = '1';
		wait until rising_edge(clk);
		ack <= '1', '0' after CLK_PERIOD + 1 ps;
	end process;

	clk <= not clk after CLK_PERIOD / 2;

end architecture;

#>> pkg.vhd
package sim_types_pkg is
    type descriptor_t is record
		address : natural;
		length : positive;
	end record;
end package;

#>> run.py
from vunit import VUnit

vu = VUnit.from_argv()
vu.add_verification_components()
lib = vu.add_library('lib')
lib.add_source_files('dma_controller_tb.vhd')
lib.add_source_files('sim_types_pkg.vhd')
pkg = lib.package('sim_types_pkg')
pkg.generate_codecs('sim_codecs_pkg')
vu.set_compile_option('ghdl.flags', ['-g'])
tb = lib.test_bench('dma_controller_tb')
vu.main()

#>> sim.sh
python run.py

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Checklist**
Reproducer archived:
[ghdl.zip](https://github.com/ghdl/ghdl/files/2633566/ghdl.zip)

Before submitting your issue, please review the following checklist:

- [ ] Add `GHDL Bug occurred` log block
- [ ] Add a MWE
- [x] Try the latest version/home

tgingold, 2018-12-02:
I can reproduce the issue.

Paebbels, 2018-12-02:
@tgingold so I assume it's a bug, right? (added Bug label).

tgingold, 2018-12-03:
But as @Paebbels said on gitter, I am not sure that a function call is allowed as a prefix of 'range.

According to LRM08 6.4, a return value is not an object.
And according to 16.2.3, the prefix is appropriate for an array object (or an alias or designates an array subtype).

go2sh, 2018-12-03:
I think this is allowed, when you go with LRM08 Chapter  8. A prefix can be a name or a function call. With the following said:
>  If the prefix of a name is a function call, then the name denotes an element, a slice, or an attribute, either of the result of the function call, or (if the result is an access value) of the object designated by the result.
A attribute name just requires a prefix.

tgingold, 2018-12-03:
I don't agree with 'just'.  It requires a prefix but there are additional restrictions given in 16.2.
For example, you cannot use a name of an entity as a prefix for 'range.

slaweksiluk, 2018-12-03:
I hope this is allowed as it is convenient  to hide complex range expressions. In general it would be great to have function return range explicitly in vhdl like:
```vhdl
function get_range return range is begin
return 0 to 1;
end function;
```
```vhdl
for i in get_range loop
end loop;
```
But I suspect it would not be so easy to define it formally.

Paebbels, 2018-12-03:
@slaweksiluk what you're requesting is part of VHDL-2018 :).

@tgingold, @go2sh 
I tested with Aldec's Riviera-PRO 2018.02. It's accepting the following *minimal* code:

```vhdl
entity e is
end entity;

architecture a of e is
	function foo(n : positive := 3) return bit_vector is
	begin
		return (n downto 0 => '0');
	end function;
begin
	process
	begin
		for i in foo'range loop
			report integer'image(i);
		end loop;
		
		for i in foo(2)'reverse_range loop
			report integer'image(i + foo'length);
		end loop;
		
		wait; --forever
	end process;
end architecture;
```

It is compiling with `-2008` flag, simulating and writing the outputs: 3,2,1,0,4,5,6
The code tests:
* `'range`
* `'reverse_range`
* `'length`

whereby the function is called with and without a parameter.

Paebbels, 2018-12-04:
So, some LRM reading ...

For `A'range[(N)]`, the prefix is defined as:
> Any prefix A that is appropriate for an array object, or an alias thereof,  
> or that denotes an array subtype whose index ranges are defined by a  
> constraint.

So, is a function call *appropriate* or a functions return value *appropriate* for an array object?

**Definition of `appropriate`**
> A prefix is said to be appropriate for a type if the type of the prefix is the type considered, or if  
> the type of the prefix is an access type whose designated type is the type considered. (8.1)

Reading further in 8.1 Names - General...

    prefix ::=
        name
      | function_call

> Certain forms of name (indexed and selected names, slice names, and attribute names) include a prefix that is a name or a **function call**. **If the prefix of a name is a function call, then the name denotes an element, a slice, or an attribute, either of the result of the function call, or (if the result is an access value) of the object designated by the result.** Function calls are defined in 9.3.4.
> 
> A prefix is said to be appropriate for a type in either of the following cases:
> * The type of the prefix is the type considered.
> * The type of the prefix is an access type whose designated type is the type considered.

Section 8.6 Attribute names doesn't define any restrictions for a function call as a prefix.

So I believe @go2sh and Aldec are correct.

tgingold, 2018-12-04:
The problem is not about array, but about object. A function call is not an object.
It would be ok if it were written: appropriate for an array value.

Paebbels, 2018-12-04:
@ JimLewis any ideas about this discussion and the question about function calls as a prefix for `'range`?

Can anyone test modelsim?

tmeissner, 2018-12-04:
@Paebbels I can run your testcase with Modelsim later at work.

go2sh, 2018-12-04:
@Paebbels With 10.7c it works.

@tgingold I agree, that not every prefix is suitable for every attribute. But this would be a problem of the semantics not the syntax, which I was referring to. Also, I think 16.2.3 allows it:

>Any prefix A that is appropriate for an array object, or an alias thereof, or that denotes an array subtype  whose index ranges are defined by a constraint.

Especially the last part: "Any prefix A ... that denotes an array subtype... ". I think this is directly referring to the return value of a function call.

JimLewis, 2018-12-04:
@Paebbels @tgingold  Given that it works on other simulators, I would think less about what you think the LRM says and think more about if this is a valuable feature to the language or not.   If the wording in the LRM is ambiguous, we can fix it in the next revision if necessary.   Not every deviation from the LRM done by commercial vendor is unintentional - some are customer requests.

tgingold, 2018-12-04:
@go2sh, we agree about the syntax.  But the semantics is also important.
A return value is not a subtype, and a function call doesn't denote.

tgingold, 2018-12-04:
@JimLewis that's not really an argument for the standardisation process!

closed 2018-12-06 by tgingold

JimLewis, 2018-12-07:
@tgingold  IEEE Standards are done by consensus.   If the community agrees that a feature is valuable it is considered for inclusion in the language - given the use models agree.    In this situation, you have a community who wants a feature, you have vendors already implementing it that way, unless there are compelling reasons and/or use models not to, doesn't it make sense to clarify the LRM to make it so.

tgingold, 2018-12-11:
The main drawbacks of this approach are:
- different vendors may implement different features, or make different implementations; that creates incompatibilities
- the standard doesn't anymore define the language (so why would we need a standard)