ghdl/ghdl #714: How do I use the sdf annotation?

doppioandante, 2018-11-30:
Hi, It's not clear to me how to use the --sdf flag from the docs...
I have the following files:
* post_route.vhd (top entity: synth_top)
* post_route.sdf
* synth_top_tb.vhd (top entity: synth_top_tb)

`post_route.vhd` and `post_route.sdf` are generated by Xilinx Vivado, and I can run the testbench successfully, but I can't figure how to tell ghdl that synth_top should be annotated with the SDF file... what is the correct usage?

tgingold, 2018-12-01:
It should be something like `--sdf=synth_top_tb.synth_top=post_route.sdf`

There are very few ghdl users of sdf.

doppioandante, 2018-12-02:
> It should be something like `--sdf=synth_top_tb.synth_top=post_route.sdf`
> 
> There are very few ghdl users of sdf.
Thanks, I figured it out now. Basically, i need to do `--sdf=name_of_the_instance=sdf_file.sdf`

Unfortunately ghdl doesn't understand `PATHPULSE`, how hard would it be to implement it? It's not a priority for me, if it turns out to be easy conceptually I could dig into the SDF standard and ghdl's code in a few months

tgingold, 2018-12-02:
You can continue in this issue.  `PATHPULSE` could easily be ignored in the SDF parser (see `grt/grt-sdf.adb`).