ghdl/ghdl #719: Name conflict of record elements and port / signal names

Fixitfetish, 2018-12-25:
An element of a record cannot have the name as ports and signals declared in the same entity / architecture ... same for procedures. 

test_tb.vhdl:19:5:warning: declaration of "rst" hides signal "rst" [-Whide]
test_tb.vhdl:24:14:warning: declaration of "din" hides signal "din" [-Whide]
test_tb.vhdl:25:14:warning: declaration of "dout" hides port "dout" [-Whide]
test_tb.vhdl:24:14:warning: declaration of "din" hides signal "din" [-Whide]
test_tb.vhdl:25:14:warning: declaration of "dout" hides port "dout" [-Whide]
ERROR: Compilation failed.
 
This bug has been introduced with 0.36-dev (20181129) but was working correctly with 0.35.

Code to reproduce :
----------------------------------------------------------------------------------------------
 library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;
entity test_tb is
port(
  dout : std_logic
);
end entity;

architecture sim of test_tb is
  
  signal rst : std_logic := '1';
  signal clk : std_logic := '1';
  signal din : std_logic ;

  type myrec is
  record
    rst : std_logic;
    vld : std_logic; 
  end record;

  procedure myproc(
    variable din : in std_logic;
    variable dout : out std_logic
  ) is
  begin
    dout := din;
  end procedure;

begin
end architecture;

Paebbels, 2018-12-25:
Your analyze log shows mainly warnings. Currently, I cannot see why the compilation stopped for your code (and where...).

The warning about `rst` shouldn't be a warning, because subelement names in a type cannot hide objects. They are no objects names yet. The doubled warning about `din`and `dout`
 should be improved to occur only once.

The warning itself about the variables `din` and `dout` hiding the declarations of a port and a signal are valid.

Fixitfetish, 2018-12-26:
Thanks for the quick reaction! 
@Paebbels : I did not get your last point why "The warning itself about the variables din and dout hiding the declarations of a port and a signal are valid."  Aren't the variables `din` and `dout` local to the procedure?

Paebbels, 2018-12-26:
@Fixitfetish if you don't declare the two variables in the procedure, then `dout <= din;` is still valid. (Please not I changed from variable assignment to signal assignment.) The signals and ports (which are also signals) are declared in an outer scope compared to the procedure. The variable declarations hide these names.