ghdl/ghdl #720: LLVM support

Defossez, 2018-12-26:
Hi,

Compiling GHDL with LLVM only (following the on line doc) supports LLVM version 3.5 till 5.0.
LLVM is already at versions 6.0 and 7.0 (Ubuntu 18.04).

Will GHDL compilation with LLVM 6.0 or 7.0 work?
LLVM version 3.5 to 5.0 are listed/supported because they have been tested.
Or will a GHDL compilation with LLVM 6.0 or 7.0 jsut not work at all?

Asking the question before starting anything.

Thanks,

1138-4EB, 2018-12-28:
ATM, GHDL is tested with LLVM 3.8 (Ubuntu 14), 3.9 (Ubuntu 16), 4.0.1 (Fedora 26), 5.0 (Ubuntu 18), 5.0.2 (Fedora 27) and 6.0.1 (Fedora 28). See: https://travis-ci.com/ghdl/docker/builds/95855133

Defossez, 2018-12-31:
OK thanks.
I'll tried it on Ubuntu 18.04 with LLVM 6.0 / 7.0 and GNAT 2018-20180528 (latest version from web site).
It works perfectly.