ghdl/ghdl #723: Top-most testsuite does not pass options to individual testsuits

Paebbels, 2018-12-29:
**Description**
Test suits like `sanity` support parameters like `-k` for *keep going*.  
This option cannot be set from top-most test suite file.

**Expected behavior**
Allow setting of global (useful) options to all test suits from top-most test suite.

------------
Is this code incomplete?

```bash
for opt; do
  case "$opt" in
      [a-z]*) tests="$tests $opt" ;;
      *) echo "$0: unknown option $opt"; exit 2 ;;
  esac
done
```

... because, `$opt` is not declared, thus it's empty.

Paebbels, 2019-01-05:
For the code above: I figured out that `for opt;` is reading from `$@`.

As I was investigating #724, it needed some extra effort to run the suits or individual tests independently with and without `-k`. Especially when waiting >30 minutes until one stopped or finished.

I would prefer to have the `-k` option also on the top-most testsuite.

tgingold, 2019-01-06:
I don't understand your point.

The vests suite is not very interesting.  It tests most of the 93 features, but those are implemented.  The vests suite rarely fails.

The sanity suite is very small and very simple.

The gna suite (in fact the regression test suite) is the most interesting one.  In fact, I run only this suite during development.  It supports -k, and to run one test just go in its directory and execute ./testsuite.sh  There is also the testsuite.py driver (linux only - to be ported on windows) to run it using multiple cpus.

PR #1615 (2021-01-18): testsuite: add suite_driver.sh
Close #723

`testsuite/gna/testsuite.sh`, `testsuite/sanity/testsuite.sh`, `testsuite/synth/testsuite.sh` and `testsuite/vpi/testsuite.sh` are almost identical. In this PR, `testsuite/suite_driver.sh` is created, which takes the target suite name as the first argument. Then, those four testsuites are rewritten for calling the suite_driver. At the same time `testsuite/testsuite.sh` is updated for calling the suite_driver directly. BTW, positional arguments to be passed to the suite_driver are supported:

```sh
# All test suites, no opts
./testsuite.sh

# Test suit [... cut]
closed 2021-01-18 by tgingold