ghdl/ghdl #724: Failing sanity checks on MinGW64

Paebbels, 2018-12-29:
**Description**
In preparation to debug failing tests on AppVeyor, I tried to run the tests in my MinGW64 environment (GCC 8.2.1, G++ 8.2.1, CLANG 7.0.1, LLVM 7.0.1). GHDL is compiled with LLVM backend and libbacktrace.

The first failing test is `sanity/000hello`.
`ghdl --elab-run` exits without any notice and gives a Bash exitcode of `127` (-1).

Also running GHDL elaboration and run (-e / -r) individually, behaves in the same way.

When I execute the created executable `hello.exe` I get a segmentation fault:
```
Patrick Lehmann@PLC2-Paebbels MINGW64 /c/git/GitHub/Paebbels/ghdl/testsuite/sanity/000hello
$ ./hello.exe
Segmentation fault
```

Running `hello.exe` in GDB reports this:
```
Patrick Lehmann@PLC2-Paebbels MINGW64 /c/git/GitHub/Paebbels/ghdl/testsuite/sanity/000hello
$ gdb hello
GNU gdb (GDB) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from hello...done.
(gdb) r
Starting program: C:\git\GitHub\Paebbels\ghdl\testsuite\sanity\000hello\hello.exe
[New Thread 29528.0x6eac]
[New Thread 29528.0x55f8]
[New Thread 29528.0x7304]
[New Thread 29528.0x65c8]

Thread 1 received signal SIGSEGV, Segmentation fault.
0x0000000000409132 in <__ghdl_rti_add_package> (pkg=0x88ad040)
    at C:\git\GitHub\Paebbels\ghdl\src\grt\grt-rtis.adb:32
32            Ghdl_Rti_Top.Children (Ghdl_Rti_Top.Nbr_Child) := Pkg;
(gdb)
```

Does this already help or can I do further debug steps for you?  
Please advice what (and how :) ) to debug with GDB.

Paebbels, 2018-12-29:
This is the stack trace (looks so short ...)

```
(gdb) r
Starting program: C:\git\GitHub\Paebbels\ghdl\testsuite\sanity\000hello\hello.exe
[New Thread 16500.0x7994]
[New Thread 16500.0x7370]
[New Thread 16500.0x44dc]
[New Thread 16500.0x66b0]

Thread 1 received signal SIGSEGV, Segmentation fault.
0x0000000000409132 in <__ghdl_rti_add_package> (pkg=0x8871040)
    at C:\git\GitHub\Paebbels\ghdl\src\grt\grt-rtis.adb:32
32            Ghdl_Rti_Top.Children (Ghdl_Rti_Top.Nbr_Child) := Pkg;
(gdb) where
#0  0x0000000000409132 in <__ghdl_rti_add_package> (pkg=0x8871040)
    at C:\git\GitHub\Paebbels\ghdl\src\grt\grt-rtis.adb:32
#1  0x00000000004015d7 in __ghdl_ELABORATE ()
#2  0x00000000004b0500 in work__hello__ARCH__behav__INSTSIZE ()
#3  0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
(gdb) list
27         Max_Top_Pkg : Ghdl_Index_Type := 0;
28
29         procedure Ghdl_Rti_Add_Package (Pkg : Ghdl_Rti_Access) is
30         begin
31            pragma Assert (Ghdl_Rti_Top.Nbr_Child < Max_Top_Pkg);
32            Ghdl_Rti_Top.Children (Ghdl_Rti_Top.Nbr_Child) := Pkg;
33            Ghdl_Rti_Top.Nbr_Child := Ghdl_Rti_Top.Nbr_Child + 1;
34         end Ghdl_Rti_Add_Package;
35
36         procedure Ghdl_Rti_Add_Top (Max_Pkg : Ghdl_Index_Type;
(gdb) info args
pkg = 0x8871040
(gdb) info locals
No locals.
(gdb)
```

Paebbels, 2018-12-29:
The same behavior can be reproduced with `gna/bug01`:

```
Patrick Lehmann@PLC2-Paebbels MINGW64 /c/git/GitHub/Paebbels/ghdl/testsuite/gna/bug01
$ gdb foo.exe
GNU gdb (GDB) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from foo.exe...done.
(gdb) r
Starting program: C:\git\GitHub\Paebbels\ghdl\testsuite\gna\bug01\foo.exe
[New Thread 20580.0xf14]
[New Thread 20580.0x6210]
[New Thread 20580.0x63c0]
[New Thread 20580.0x7acc]

Thread 1 received signal SIGSEGV, Segmentation fault.
0x0000000000425ab2 in <__ghdl_rti_add_package> (pkg=0x8969040)
    at C:\git\GitHub\Paebbels\ghdl\src\grt\grt-rtis.adb:32
32            Ghdl_Rti_Top.Children (Ghdl_Rti_Top.Nbr_Child) := Pkg;
(gdb) where
#0  0x0000000000425ab2 in <__ghdl_rti_add_package> (pkg=0x8969040)
    at C:\git\GitHub\Paebbels\ghdl\src\grt\grt-rtis.adb:32
#1  0x00000000004015d7 in __ghdl_ELABORATE ()
#2  0x00000000004d0e50 in work__foo__ARCH__fum__INSTSIZE ()
#3  0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
(gdb) info args
pkg = 0x8969040
(gdb) info locals
No locals.
(gdb)
```

Paebbels, 2018-12-29:
I tried to check AppVeyor when the problem got introduced, but this problem got covered by the GCC / LLVM version problem that I fixed with one of the last commits.

The problem got introduced between cd8247bb (last positive AppVeyor run) and 5bf6c7e0 (next AppVeyor run that successfully compiled again). See diff: cd8247bb...5bf6c7e0

@1138-4EB is this a case where we could use `git bisect` to narrow down 182 commits?

Paebbels, 2018-12-29:
Modifications to `grt-rtis.adb` where made here:
* d4708cc3 (24.11.2018) trans-rtis: take into account instances in the count of packages.  
  rtis: check number of packages.
* ed7ad157 (17.10.2018) Rework translation of unbounded and complex types.

According to the commit messages for *rtis* related changes, it is related to unconstrained records or rework done for unconstrained records.

tgingold, 2018-12-29:
> The problem got introduced between cd8247b (last positive AppVeyor run) and 5bf6c7e (next AppVeyor run that successfully compiled again). See diff: cd8247b...5bf6c7e

I am not sure the bug was introduced by a change. It is possible that it was present but was made visible by llvm 7.

tgingold, 2018-12-29:
Can you do: print ghdl_rti_top.all

Paebbels, 2018-12-29:
That's also possible.

MinGW allows me to install only the latest version of a program. I see no way to test GHDL with LLVM 4/5/6 in MinGW.

OTOH, I have a Debian Testing running with GCC/G++/GNAT 8.2 and LLVM/CLANG 7.0 too. I can try to execute the test suite there too. This might harden the suspicion against a newer LLVM version.

Paebbels, 2018-12-29:
My Linux machine reports:
```
paebbels@Runner1:~/git/ghdl/testsuite/sanity/000hello$ GHDL=/opt/ghdl/0.36-dev-gcc82-llvm70/bin/ghdl ./testsuite.sh
analyze hello.vhdl
elaborate and simulate hello
hello.vhdl:6:3:@0ms:(assertion note): Hello VHDL world
Remove work library
test successful
```

So it's MinGW64 and/or Windows related?

Paebbels, 2018-12-29:
I extended the gdb output log 2 comments above. `print ghdl_rti_top.children(0)` is an access value pointing to address `0x105`, which is on page 0, where no access is allowed.

Does Ada support pointer arithmetics?  
How was `0x105` generated?  
Looks like a base address is missing.

Paebbels, 2018-12-31:
So, what could be a solution?

tgingold, 2018-12-31:
Using the new function to normalize a triple fixes the issue.

closed 2018-12-31 by tgingold

Paebbels, 2019-01-01:
I believe not everything is fixed with this commit...

* AppVeyor
  * gna tests are all positive
  * vests tests are disabled
  * sanity tests are not included in AppVeyor runs
* My local MinGW64
  * gna tests have one failing:
    * issue394
      ```
      GNA dir issue394:
      analyze bug.vhdl
      Dateien out.txt und out.ref sind verschieden.
      ```
      Both files are different: This is the result in MinGW64:
      ```
      elaborate and simulate bug
      /opt/ghdl/0.36-dev/bin/ghdl --elab-run    bug
      1
      bug.vhdl:18:5:@0ms:(report note): 2
      3
      bug.vhdl:20:5:@0ms:(report note): 4
      ```
      This was expected:
      ```
      elaborate and simulate bug
      1
      bug.vhdl:18:5:@0ms:(report note): 2
      3
      bug.vhdl:20:5:@0ms:(report note): 4
      ```
  * vests tests are all positive
  * sanity checks have one failing:
    * 005examples
      ```
      sanity 005examples:
      Extracting hello.vhdl...
      #################################################################
      ######### FAILURE: 005examples
      #################################################################
      sanity test failed ( 005examples)
      ```

      Problem: The extracted `hello.vhdl` has 0 bytes.

tgingold, 2019-01-02:
Vests tests are not run on Appveyor as they take too much time.

for issue394, it is a little bit strange you get the command line in the output.
I will try.

For 005examples, I suspect a \n vs \r\n issue.

tgingold, 2019-01-02:
On my msys2-w64 system, 005examples work correctly with both python2 and python3.
Which python are you using (both version and native/msys2) ?

Paebbels, 2019-01-02:
Because I have quite a lot of Pythons on my machine and my environments, I tried to discover all of them. It will also list the versions for each installation. `which python` in MinGW63 confirms, that the Python 3.7.2 from MSYS2 is used.

**My MSYS2 is configured as follows:**

List of installed python packages:
```bash
$ pacman -Qqe | grep py
python
python3-pip
python3-setuptools
```

These are the versions of the mentioned packages:
```bash
$ pacman -Q | grep python
python 3.7.2-1
python3-pip 18.1-1
python3-setuptools 40.5.0-1
```

So, my MSYS2 environment has only python installed, no MinGW32 or MinGW64 variant in parallel. The version is 3.7.2. This is also confirmed by `python --version` and `which python`.

I also checked that the referenced documentation files is available and can be read e.g. by `cat`.

The Windows 10 installation has a Python 3.7.0 (64 bit) as default Python:
```powershell
PS> py.exe -0
Installed Pythons found by C:\Windows\py.exe Launcher for Windows
 -3.7-64 *
 -3.6-64
PS> py.exe -3.7 --version
Python 3.7.0
PS> py.exe -3.6 --version
Python 3.6.5
```

I have banned Python 2 from my installations :).