ghdl/ghdl #726: support of generic types (vhdl2008)

Stefichen5, 2018-12-31:
**Is your feature request related to a problem? Please describe.**
I can not successfully use entities, that have a generic type. They do analyze but when instantiating them in another entity (let's say in a testbed) I will get errors like these (example code see on the bottom):

> ghdl -a --std=08 ea-GenericMux.vhd
> ghdl -a --std=08 tbMux.vhd
> tbMux.vhd:24:32: can't associate 'ia_ulogic' with port "ia"
> tbMux.vhd:24:32: (type of 'ia_ulogic' is std_ulogic)
> ea-GenericMux.vhd:8:16: (type of port "ia" is gamuxtype)

**Describe the solution you'd like**
VHDL 2008 allows the usage of generic types. those can be used within an entity instead of actual types like std_ulogic, natural, ...
It would be great if GHDL could support them.

**Additional context**
I have successfully tested my code with Modelsim, as well as run a successful synthesis with Quartus Pro 18.1.
I have attached a simple example that demonstrates the behaviour.

[ea-GenericMux.vhd.txt](https://github.com/ghdl/ghdl/files/2718283/ea-GenericMux.vhd.txt)
[tbMux.vhd.txt](https://github.com/ghdl/ghdl/files/2718284/tbMux.vhd.txt)

Paebbels, 2018-12-31:
Generic interface types in entities are not yet supported by GHDL. It's a missing VHDL-2008 feature.

Stefichen5, 2018-12-31:
Sorry. I meant to post this as a feature request. I forgot to put on the label.
Thank you for adding it for me.

eine, 2021-01-17:
FTR,

```
# ghdl -a --std=08 ea-GenericMux.vhd 
translate_subtype_definition: cannot handle IIR_KIND_INTERFACE_TYPE_DEFINITION (ea-GenericMux.vhd:9:22)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (0.37.0.r1270.gb65ee8c3) [Dunoon edition]
Compiled with GNAT Version: 10.2.0
Target: x86_64-w64-mingw32
D:\data-dev\umarcor\ghdl\ghdl\temp\
Command line:
C:\msys64\mingw64\bin\ghdl1-llvm.exe --std=08 -PC:\msys64\mingw64\lib/ghdl\ieee\v08\ -PC:\msys64\mingw64\lib/ghdl\ -c -o ea-GenericMux.o ea-GenericMux.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x4a78bf 0x588f91 0x589575 0x5f7fba 0x5ffcf8 0x5e4b6f 0x5d7c76 0x637218 0x638bba 0x63a495 0x402706 0x4013bf 0x4014f4 0x7ffaa4177032 0x7ffaa549d0cf
******************************************************************
C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

NicoPy, 2021-10-05:
Any schedule for a fix ?

tgingold, 2021-10-19:
Generic types are supported for packages, but neither for subprograms nor for entities.

It will come in the future.

closed 2024-01-02 by tgingold

Galactic-Explorer, 2024-11-07:
Is this but fixed? because I got GHDL Bug when I tried to run the example provided.
`******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.1.0.r281.g282567c96) [Dunoon edition]
Compiled with GNAT Version: 14.1.0
Target: x86_64-w64-mingw32
C:\dev\ghdl-bugs\
Command line:
C:\bin\msys64\mingw64\bin\ghdl1-llvm.exe --std=08 -PC:\bin\msys64\mingw64\lib/ghdl\ieee\v08\ -PC:\bin\msys64\mingw64\lib/ghdl\ -c -o tbMux.o tbMux.vhd
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap9.adb:2305 access check failed
Load address: 0x7ff750690000
Call stack traceback locations:
0x7ff750aa7bf4 0x7ff750aa39bd 0x7ff750aa42f8 0x7ff750a58062 0x7ff750a49c4e 0x7ff750abf011 0x7ff750ac0686 0x7ff750ac1a00 0x7ff750693244 0x7ff7506912ec 0x7ff750691404 0x7ffb2a43257b 0x7ffb2b3aaf06
******************************************************************`  

Windows 11 Build from source: 282567c96a05bbc8eec45fcf20429ccec74ce1d5

tgingold, 2024-11-08:
It works with the mcode backend, but not handled by the llvm backend.

Galactic-Explorer, 2024-11-18:
Will it be supported with the LLVM backend?

tgingold, 2024-11-19:
Do you have any strong reason to use the llvm back-end ?

It is not an llvm issue (it will work with llvm when configured with --with-llvm-jit), but jit vs not jit backend.  And by default, llvm is not jit.

creiter64, 2024-11-19:
I'd also be interested in getting this to work on the llvm backend.

As a side note, what does `--with-llvm-jit` do and what is the difference to the "normal" LLVM backend? I couldn't find any information regarding that flag in the docs: https://ghdl.github.io/ghdl/development/building/LLVM.html

tgingold, 2024-11-19:
Indeed, this flag is not yet documented.

With the non-jit backends, code is generated when a file is analyzed.  This works until vhdl-2008 as it is not easy to generate code for generic types.

with the jit backends, code is generated once the whole design is elaborated.

By default, llvm is a non-jit backend.  But mcode is always a jit backend.

commit 3f772322 (2024-01-02): vhdl: improve support of type interfaces in entities

Fix #726