ghdl/ghdl #728: List all warning that are enabled by default

Paebbels, 2019-01-03:
I was searching the documentation on how to disable the `-Whide` warning. I noticed, that there is no list of warning, which are enabled by default.

Please add this list to the section about warnings: https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html?highlight=disable#warnings

This is what I figured out so far:
* `-Whide`
* `-Wothers`

Is there a different set of warnings enabled, when GHDL runs in verbose mode (`-v`)?

closed 2019-01-06 by tgingold

Paebbels, 2019-01-06:
@tgingold accordig to what I figured out while running the pre-compile scripts, the warnings

- `Wothers`
- `Wparenthesis`
- `Wbody`
- `Wstatic`

are also enabled, but not listed in your commit 0105520.

tgingold, 2019-01-06:
All warnings were enabled before.

commit 01055202 (2019-01-06): Document warnings enabled by default.

Close #728