ghdl/ghdl #731: GHDL elab-run error when instanciating entity

Stefichen5, 2019-01-05:
**Description**
I wanted to write a simple testbench for a simple adder. The adder uses VHDL 2008's slices in aggregates feature. GHDL can analyze the adder and testbench without problems. When running 
> ghdl --elab-run --std=08 tbAdder

I get a GHDL error. I stripped the testbench down. It is now just the instantiation of the adder but the error stays.

**Expected behaviour**
The elab-run should not crash.
I verified the correctness and functionality of my adder as well as the (full) testbench with Modelsim (adder + testbench) as well as Quartus Prime Pro 18.1 (just the adder with a separate testbed)

**Context**
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (20181129) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: i686-w64-mingw32
In directory: C:\Users\stefa\BAKThesis\VHDL\SlicesInAggregates\
Command line:
C:\Users\stefa\GHDL\20181129-mingw32-mcode\bin\ghdl.exe --elab-run --std=08 tbAdder
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap8.adb:4312
Call stack traceback locations:
0x632398 0x636bbe 0x69b0a2 0x61260a 0x610012 0x61afde 0x69e23d 0x5f8dc6 0x58ee78 0x6a0623 0x402635 0x401389 0x7660fe07 0x76f1662b 0x76f165fb
******************************************************************

- OS: Win10
- Origin:
  - [ ] Package manager. Repo:
  - [X] Released binaries. Tarball: 20181129
  - [ ] Built from sources. Commit SHA:
- Version: GHDL 0.36-dev (20181129)
**Additional context**
I will attach a stripped down version of the files I used to this bug report.

**How to reproduce?**
Download my files, analyze them and run --elab-run

```
#>> ea-Adder.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity Adder is
	generic (gWidth : natural);
	port(
		iA : in std_ulogic_vector(gWidth-1 downto 0);
		iB : in std_ulogic_vector(gWidth-1 downto 0);
		
		oCarry: out std_ulogic;
		oRes : out std_ulogic_vector(gWidth-1 downto 0)
	);
end entity Adder;

architecture RTL of Adder is
begin
	(oCarry, oRes) <= std_ulogic_vector(unsigned('0' & iA) + unsigned('0' & iB));
end architecture RTL;

#>> tbAdder.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.env.all;

entity tbAdder is
end entity tbAdder;

architecture Bhv of tbAdder is
	constant cWidth : natural := 8;	
	signal iA, iB : std_ulogic_vector(cWidth-1 downto 0) := (others => '0');
	signal oRes : std_ulogic_vector(cWidth-1 downto 0);
	signal oCarry : std_ulogic;
begin
	UUT: entity work.Adder
		generic map(
			gWidth => cWidth
		)
		port map(
			iA     => iA,
			iB     => iB,
			oCarry => oCarry,
			oRes   => oRes
		);	
end architecture Bhv;

#>> sim.sh
ghdl -a --std=08 ea-Adder.vhd
ghdl -a --std=08 tbAdder.vhd
ghdl --elab-run --std=08 tbAdder

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Files**
[ea-Adder.vhd.txt](https://github.com/ghdl/ghdl/files/2729249/ea-Adder.vhd.txt)
[tbAdder.vhd.txt](https://github.com/ghdl/ghdl/files/2729250/tbAdder.vhd.txt)

**Checklist**
Before submitting your issue, please review the following checklist:

- [X] Add `GHDL Bug occurred` log block
- [X] Add a MWE
- [X] Try the latest version

tgingold, 2019-01-05:
No crash with the current ghdl on the master branch.  I think this bug has been fixed recently so you simply need to update.

Thank you for the report.

closed 2019-01-05 by tgingold

Stefichen5, 2019-01-05:
Thank you. I just pulled the sources and compiled them. Works now. I am just having this issue with the latest prebuilt binary.