ghdl/ghdl #736: VHDL 2008 conditional assignment (when-else) inside a combinational process incorrect behaviour

xnorme, 2019-01-07:
**Description**
I'm trying to come up with a MWE, but what I have is a statement like:
```
a <= "0000" when c = 0 else (d - 1) & e + 8;
```
inside a `process(all)` which was assigning an incorrect value to a. Moving the exact same assignment outside the process, without touching anything else, results to correct behavior.

This is a big (and proprietary) design but I'll try to come up with a minimal example as soon as I can.

**Expected behaviour**
What you expected to happen, and what is happening instead.

**Context**
Please paste the `GHDL Bug occurred` log block here. Also, provide the following information:

- OS: MacOS
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: a29a5283

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE), that is compatible with [issue-runner](https://github.com/1138-4EB/issue-runner). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```
#>> ent.vhd
entity ent is
end entity;

architecture a of ent is
begin
  process begin
    report "Hello world" severity note;
    wait;
  end process;
end;

#>> sim.sh
ghdl -a ent.vhd
ghdl --elab-run ent

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

Note that `run.sh` is used to execute `sim.sh` inside a docker container. Please, put your commands in `sim.sh` and just copy `run.sh` from the example. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

**Files**
A list of relevant files for this issue. Large files can be uploaded one-by-one or in a tarball/zipfile. See [1138-4EB/issue-runner#parser](https://github.com/1138-4EB/issue-runner#parser).

**Checklist**
Before submitting your issue, please review the following checklist:

- [ ] Add `GHDL Bug occurred` log block
- [ ] Add a MWE
- [x] Try the latest version

Paebbels, 2019-01-07:
Can you provide a minimal VHDL example for your case?

tgingold, 2019-01-08:
So there was no crash, but an incorrect result.

I gave a quick look at the code for conditional signal assignment statement, but I have found nothing suspicious.

tgingold, 2019-03-15:
Ok, I can reproduce the issue.  Investigating.

tgingold, 2019-03-16:
Thank you for the reproducer, that was really useful.  I will push the fix soon.

commit 6e8e0f9d (2019-03-16): vhdl: do not clear original node in canon for conditional signal assignment.

Fix #736
closed 2019-03-16 by tgingold