ghdl/ghdl #737: Unconstrained array assignement causes ORTHO_CODE.SYNTAX_ERROR

FenrisTC, 2019-01-09:
**Description**
Assigning a concatenated expression to a signal of an unconstrained array of unconstrained arrays crashes GHDL during elaboration, analysis runs fine.

**Context**
```sh
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (20181129-154-g5a74fc46) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: x86_64-linux-gnu
In directory: /tmp/bug/
Command line:
ghdl -c --std=08 -e ent
Exception ORTHO_CODE.SYNTAX_ERROR raised
Exception information:
raised ORTHO_CODE.SYNTAX_ERROR : ortho_code-exprs.adb:1063
Call stack traceback locations:
0x55ba09748c8a 0x55ba0975fea7 0x55ba098001e4 0x55ba098efb5e 0x55ba098e5b2a 0x55ba098e5246 0x55ba098e5384 0x55ba098e52cf 0x55ba098e4ffb 0x55ba098e7fe4 0x55ba098efc57 0x55ba098ef54e 0x55ba098ba84e 0x55ba098ba979 0x55ba098bafb8 0x55ba098bb6f3 0x55ba098bcdc4 0x55ba098bcef7 0x55ba098c036d 0x55ba098c4bf9 0x55ba09930402 0x55ba0989c2ed 0x55ba098994b9 0x55ba098a71cb 0x55ba0993365d 0x55ba0987d50a 0x55ba09806316 0x55ba099363c8 0x55ba0963c6c3 0x7fc16a70ab95 0x55ba0963b4a8 0xfffffffffffffffe
******************************************************************
```

- OS: Ubuntu 18.04
- Origin:
  - [x] Built from sources. Commit SHA: 5a74fc46

**How to reproduce?**

```
#>> ent.vhd
entity ent is
end entity;

architecture impl of ent is
	type bitvv is array(natural range <>) of bit_vector;
	signal foo: bitvv(2 downto 0)(0 downto 0);
begin

foo <= foo(1 downto 0) & foo(2);

end architecture;

#>> sim.sh
ghdl -a --std=08 ent.vhd
ghdl --elab-run --std=08 ent

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Additional context**
I tried out different ways to do this assignment to find a quick workaround and I found multiple distinct crash reports.
If it seems helpful, I can post MWEs together with the reports either here or as different issues.

FenrisTC, 2019-01-09:
Alright, here you go.

```vhdl
entity ent is
end entity;

architecture impl of ent is
	type bitvv is array(natural range <>) of bit_vector;
	signal foo: bitvv(2 downto 0)(0 downto 0);
begin

foo <= (
	2 downto 1 => foo(1 downto 0),
	0 => foo(2)
);

end architecture;
```
```sh
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (20181129-154-g5a74fc46) [Dunoon edition]
Compiled with GNAT Version: Community 2018 (20180524-73)
Target: x86_64-pc-linux-gnu
In directory: /tmp/
Command line:
ghdl -e --std=08 ent
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap3.adb:3056 access check failed
Call stack traceback locations:
0x55cfb95771f2 0x55cfb953cc88 0x55cfb953d897 0x55cfb953cf26 0x55cfb953edc0 0x55cfb953ee9d 0x55cfb95420a7 0x55cfb9541c09 0x55cfb950b08f 0x55cfb950b1e2 0x55cfb950b8cf 0x55cfb950c07f 0x55cfb950d068 0x55cfb950d199 0x55cfb95106c0 0x55cfb95151ca 0x55cfb9583fb4 0x55cfb94ec661 0x55cfb94e9d91 0x55cfb94f6f3f 0x55cfb95873bf 0x55cfb94cf751 0x55cfb9455cf8 0x55cfb9589ee5 0x55cfb92a708c 0x7ffa61d70c05 0x55cfb92a5d78 0xfffffffffffffffe
******************************************************************
```

******************

```vhdl
entity ent is
end entity;

architecture impl of ent is
	type bitvv is array(natural range <>) of bit_vector;
	signal foo: bitvv(2 downto 0)(0 downto 0);
begin

foo <= bitvv'(
	2 downto 1 => foo(1 downto 0),
	0 => foo(2)
);

end architecture;
```

```sh
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (20181129-154-g5a74fc46) [Dunoon edition]
Compiled with GNAT Version: Community 2018 (20180524-73)
Target: x86_64-pc-linux-gnu
In directory: /tmp/
Command line:
ghdl -e --std=08 ent
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap3.adb:49
Call stack traceback locations:
0x563228d4984f 0x563228d4e26c 0x563228d4e554 0x563228d4e991 0x563228d5a78b 0x563228d24f82 0x563228d2523b 0x563228d24c09 0x563228cee08f 0x563228cee1e2 0x563228cee8cf 0x563228cef07f 0x563228cf0068 0x563228cf0199 0x563228cf36c0 0x563228cf81ca 0x563228d66fb4 0x563228ccf661 0x563228cccd91 0x563228cd9f3f 0x563228d6a3bf 0x563228cb2751 0x563228c38cf8 0x563228d6cee5 0x563228a8a08c 0x7fc04dc79c05 0x563228a88d78 0xfffffffffffffffe
******************************************************************
```

**************

```vhdl
entity ent is
end entity;

architecture impl of ent is
	type bitvv is array(natural range <>) of bit_vector;
	signal foo: bitvv(2 downto 0)(0 downto 0);
	signal bar: bitvv(1 downto 0)(0 downto 0);
begin

foo(2 downto 1) <= bar;
foo(0) <= foo(2);

end architecture;
```

```sh
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (20181129-154-g5a74fc46) [Dunoon edition]
Compiled with GNAT Version: Community 2018 (20180524-73)
Target: x86_64-pc-linux-gnu
In directory: /tmp/
Command line:
ghdl -e --std=08 ent
Exception ORTHO_CODE.SYNTAX_ERROR raised
Exception information:
raised ORTHO_CODE.SYNTAX_ERROR : ortho_code-exprs.adb:464
Call stack traceback locations:
0x562855946e0f 0x5628559489aa 0x56285595f0c7 0x5628559f20dd 0x5628559f2881 0x5628559f2c3c 0x562855ab0204 0x562855ab08cf 0x562855ab107f 0x562855ab2068 0x562855ab2199 0x562855ab56c0 0x562855aba1ca 0x562855b28fb4 0x562855a91661 0x562855a8ed91 0x562855a9bf3f 0x562855b2c3bf 0x562855a74751 0x5628559facf8 0x562855b2eee5 0x56285584c08c 0x7fda0c372c05 0x56285584ad78 0xfffffffffffffffe
******************************************************************
```

tgingold, 2019-01-11:
There are indeed an issue with slicing and an issue with concatenation.

umarcor, 2021-08-15:
FTR, 3 of the 4 reproducers above seem to be supported now:

```vhdl
entity ent is
end entity;

architecture impl of ent is
	type bitvv is array(natural range <>) of bit_vector;
	signal foo: bitvv(2 downto 0)(0 downto 0);
begin

foo <= foo(1 downto 0) & foo(2);

end architecture;
```

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r565.gb1eb0b50) [Dunoon edition]
Compiled with GNAT Version: 10.3.0
Target: x86_64-w64-mingw32
Command line:
C:\msys64\mingw64\bin\ghdl1-llvm.exe --std=08 -PC:\msys64\mingw64\lib/ghdl\ieee\v08\ -PC:\msys64\mingw64\lib/ghdl\ -c -o crash.o crash.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:1672
Call stack traceback locations:
0x7ff7963bbf10 0x7ff7964abc73 0x7ff7964ac902 0x7ff7964af0bb 0x7ff7964ad5d2 0x7ff7964ae16c 0x7ff7964df6a1 0x7ff7964de8a5 0x7ff7964de9e4 0x7ff7964de918 0x7ff7964de65c 0x7ff7964e297c 0x7ff7964edafd 0x7ff7964ed525 0x7ff79654151d 0x7ff796541682 0x7ff796541e1a 0x7ff79654289a 0x7ff796544d0f 
0x7ff796544e47 0x7ff796548c31 0x7ff79654e870 0x7ff796507fde 0x7ff7964fa49c 0x7ff79655a95d 0x7ff79655c3de 0x7ff79655dcb5 0x7ff7963226f2 0x7ff7963213af 0x7ff7963214e4 0x7ffc741c7032 0x7ffc75f8264f
******************************************************************
```

---

```vhdl
entity ent is
end entity;

architecture impl of ent is
	type bitvv is array(natural range <>) of bit_vector;
	signal foo: bitvv(2 downto 0)(0 downto 0);
begin

foo <= (
	2 downto 1 => foo(1 downto 0),
	0 => foo(2)
);

end architecture;
```

OK.

---

```vhdl
entity ent is
end entity;

architecture impl of ent is
	type bitvv is array(natural range <>) of bit_vector;
	signal foo: bitvv(2 downto 0)(0 downto 0);
begin

foo <= bitvv'(
	2 downto 1 => foo(1 downto 0),
	0 => foo(2)
);

end architecture;
```

```
crash.vhd:11:17:warning: expression subtype doesn't match aggregate element subtype [-Wruntime-error]
        0 => foo(2)
                ^
```

---

```vhdl
entity ent is
end entity;

architecture impl of ent is
	type bitvv is array(natural range <>) of bit_vector;
	signal foo: bitvv(2 downto 0)(0 downto 0);
	signal bar: bitvv(1 downto 0)(0 downto 0);
begin

foo(2 downto 1) <= bar;
foo(0) <= foo(2);

end architecture;
```

OK

commit f3e31e10 (2021-09-14): testsuite/gna: add a test and close #737
closed 2021-09-14 by tgingold