ghdl/ghdl #739: GHDL schematic viewer

rsc9421, 2019-01-15:
Hi, I'm writing a IDE for GHDL & GTKWAVE with additional features (automatically generated testbench, automatically generated declare and instance for components, etc) what do you think?
https://www.youtube.com/watch?v=e-AaoBUvnNc&t=1s
Now, I want to implement a schematic viewer (like RTL viewer in quartus or xilinx ise) and I want to ask you if I can extract any file or any source from ghdl to obtain information about conections, process (identify if is a multiplexer or a counter for example), components, etc
best regards

rsc9421, 2019-01-16:
yes, this is the link
https://github.com/rsc9421/VHDL_warrior
you can compile this project using Qt creator(binaries not avaiable yet)
how I could get hardware modules using ghdl-synth? 
i was see how to get the AST from ghdl, but I'm confused, I try using ghdl --reprint name.vhdl and I get the same file printed in console, after, I try using --file-to-xml and display much content in console, but I dont know how to extract information of these
can you help me? I'm newbie with ghdl

tgingold, 2019-01-17:
The purpose of ghdl-synth is to create a netlist; from the netlist you can extract the hardware modules.   But ghdl-synth is really WIP, so I am not sure you can go very far.

While --file-to-xml displays the AST as an XML file, it generates a very large file.  If you are interested in connections, you have to look for component_instantiation and the corresponding port_map

rsc9421, 2019-01-20:
really I tried to extract information from the xml file but, it's very hard to read 38k lines (the program becomes slowly, even in c)
thanks for help me! if you have ideas or suggestions to implement in the IDE please tell me 
regards

tgingold, 2019-01-20:
The AST is documented in the iirs.ads file; if you know VHDL, it should be quite natural to understand.

The XML dump is indeed large, but if needed you can using the python binding and extract what you need from a python program.

nuess0r, 2019-03-28:
The schematics you know from the vendor tools are generated by the synthesis tool and not from the simulation tool. Generally speaking a HDL simulator does for example not need to know what a multiplexer is.

So it is very interesting to read that there is ongoing work to use GHDL as a synthesizer :-)

If you plan to use/support Yosys in VHDL_warrior as synthesis tool, there is the command 'show' to generate schematics: http://www.clifford.at/yosys/cmd_show.html