ghdl/ghdl #740: GHDL release 0.36

tgingold, 2019-01-17:
**Description**
A clear and concise description of what the issue is about.

**Expected behaviour**
What you expected to happen, and what is happening instead.

**Context**
Please paste the `GHDL Bug occurred` log block here. Also, provide the following information:

- OS:
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [ ] Built from sources. Commit SHA:

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE), that is compatible with [issue-runner](https://github.com/1138-4EB/issue-runner). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```
#>> ent.vhd
entity ent is
end entity;

architecture a of ent is
begin
  process begin
    report "Hello world" severity note;
    wait;
  end process;
end;

#>> sim.sh
ghdl -a ent.vhd
ghdl --elab-run ent

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

Note that `run.sh` is used to execute `sim.sh` inside a docker container. Please, put your commands in `sim.sh` and just copy `run.sh` from the example. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

**Files**
A list of relevant files for this issue. Large files can be uploaded one-by-one or in a tarball/zipfile. See [1138-4EB/issue-runner#parser](https://github.com/1138-4EB/issue-runner#parser).

**Checklist**
Before submitting your issue, please review the following checklist:

- [ ] Add `GHDL Bug occurred` log block
- [ ] Add a MWE
- [ ] Try the latest version

m-kru, 2019-01-29:
When is it expected? Do we still miss some features of VHDL-2008 to compile UVVM smoothly?

tgingold, 2019-01-29:
I think it contains all the features needed to compile UVVM.

m-kru, 2019-01-30:
I have tried and it looks really promising. At the final step I get errors that GHDL can't find libraries:
`tb/irqc_tb.vhd:26:9:error: cannot find resource library "uvvm_util"`
I have compiled UVVM to `/usr/local/lib/ghdl/vendors/`, but I can't find any option to automatically provide all UVVM libraries for the build. For example something like `--lib=UVVM`.  I have read [Precompile Vendor Primitives](https://ghdl.readthedocs.io/en/latest/building/PrecompileVendorPrimitives.html), but it only describes how to compile and possible options, not how to use compiled libraries.

m-kru, 2019-01-30:
`ghdl -m --std=08 --workdir=work -P/usr/local/lib/ghdl/vendors irqc_tb` produces a set of errors like this one:
`/home/mkru/workspace/UVVM/uvvm_util/src/adaptations_pkg.vhd:65:19:error: type of a shared variable must be a protected type`

`ghdl -m --std=08 --workdir=work -frelaxed-rules -P/usr/local/lib/ghdl/vendors irqc_tb` produces no errors, however no executable file is generated.

1138-4EB, 2019-01-30:
> however no executable file is generated

Which backend are you using? If using `mcode` I think that no executable file is generated. Please, try `ghdl -r irqc_tb` after `ghdl -m...`. See second and third hint at https://ghdl.readthedocs.io/en/latest/using/QuickStartGuide.html.

m-kru, 2019-01-30:
`ghdl -r irqc_tb` produces `ghdl:error: cannot find entity or configuration irqc_tb`.
`ghdl -r --std=08 --workdir=work -frelaxed-rules -P/usr/local/lib/ghdl/vendors irqc_tb` performs the simulation, however still no executable is generated.
I am importing the sources with `-i`, then I try to build with `-m`, there are no errors however no output file is generated. When I follow the same pattern with dlx example the executable file is generated.

m-kru, 2019-01-30:
To sum up. It looks like ghdl supports UVVM.

1138-4EB, 2019-01-30:
> `ghdl -r irqc_tb` produces `ghdl:error: cannot find entity or configuration irqc_tb`.
> `ghdl -r --std=08 --workdir=work -frelaxed-rules -P/usr/local/lib/ghdl/vendors irqc_tb` performs the simulation, however still no executable is generated.

That's ok. I didn't provide all the arguments to `ghdl -r`. You do need to provide the path to the libraries.

About the backend, you can execute `ghdl --version` to see which version and backend you are using. As you now found in the docs, `mcode` won't generate an executable. However, that's ok. That's how that particular backend works. If you want to generate executables, I suggest you use LLVM as a backend.

TrampHRC, 2019-02-07:
Hi all,
looking forward to getting my hands on GHDL.
I'll wait for the 0.36 release if you say it is coming soon.
Could you roughly precise a date ? Thanks.

1138-4EB, 2019-02-07:
@TrampHRC, depending on your host OS, it might be worth either using docker images or building GHDL on your own. Certainly, you can do it now, and you will get almost (if not all) the features in release v0.36.

Availabily of prebuilt packages and/or packages in the repos of your platform might not be guaranteed.

1138-4EB, 2019-02-23:
To all of you suscribed to this issue but not watching the full repo, @tgingold just released [v0.36-rc1](https://github.com/ghdl/ghdl/releases/tag/v0.36-rc1) 🎉 🎆

1138-4EB, 2019-03-03:
Release [v0.36](https://github.com/ghdl/ghdl/releases/tag/v0.36) is ready! 🍻

closed 2019-03-10 by Paebbels